NordicSemiconductor/pc-ble-driver #1: Build error on clang-602.0.53

torleifs, 2015-06-25:
The problem seems to be that the macro EXTERN_C is not defined.
>clang --version
Apple LLVM version 6.1.0 (clang-602.0.53) (based on LLVM 3.6.0svn)
Target: x86_64-apple-darwin14.3.0
Thread model: posix

This is the result of make VERBOSE=1
[  1%] Building CXX object driver/CMakeFiles/s130_nrf51_ble_driver.dir/__/deps/nRF51_SDK_8.1.0_b6ed55f/components/serialization/application/transport/ser_softdevice_handler.c.o
cd /Users/torleifs/customers/nordic/build/driver && /usr/bin/c++   -DBLE_STACK_SUPPORT_REQD -DBOARD_NRF6310 -DHCI_LINK_CONTROL -DNRF51 -DS130 -DSD_RPC_EXPORTS -DSVCALL_AS_NORMAL_FUNCTION -Ds130_nrf51_ble_driver_EXPORTS -fPIC -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/inc -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/inc_override -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/framework/cunit/include_override -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/drivers_nrf/hal -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/drivers_nrf/uart -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/libraries/crc16 -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/libraries/scheduler -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/libraries/timer -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/libraries/util -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/serialization/application/hal -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/serialization/application/transport -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/serialization/common -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/serialization/common/transport -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/serialization/common/transport/ser_phy -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/softdevice/common/softdevice_handler -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/toolchain/gcc -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/toolchain -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/device -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/serialization/application/codecs/s130/serializers -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/serialization/common/struct_ser/s130 -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/softdevice/s130 -I/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/softdevice/s130/headers -isystem /usr/local/include    -Wall -Wno-effc++ -Wno-unknown-pragmas -Wno-undef -Wstrict-overflow=4 -Wno-long-long -Wfloat-equal -Wundef -Wshadow -Wpointer-arith -pthread -o CMakeFiles/s130_nrf51_ble_driver.dir/__/deps/nRF51_SDK_8.1.0_b6ed55f/components/serialization/application/transport/ser_softdevice_handler.c.o -c /Users/torleifs/customers/nordic/pc-ble-driver/deps/nRF51_SDK_8.1.0_b6ed55f/components/serialization/application/transport/ser_softdevice_handler.c
clang: warning: treating 'c' input as 'c++' when in C++ mode, this behavior is deprecated
In file included from /Users/torleifs/customers/nordic/pc-ble-driver/deps/nRF51_SDK_8.1.0_b6ed55f/components/serialization/application/transport/ser_softdevice_handler.c:17:
In file included from /Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/softdevice/common/softdevice_handler/softdevice_handler.h:37:
In file included from /Users/torleifs/customers/nordic/pc-ble-driver/driver/inc_override/nrf_sdm.h:17:
In file included from /Users/torleifs/customers/nordic/pc-ble-driver/driver/inc_override/nrf_soc.h:21:
/Users/torleifs/customers/nordic/pc-ble-driver/driver/../deps/nRF51_SDK_8.1.0_b6ed55f/components/softdevice/s130/headers/nrf_soc.h:374:1: error: unknown type name 'EXTERN_C'
SVCALL(SD_MUTEX_NEW, uint32_t, sd_mutex_new(nrf_mutex_t * p_mutex));

kenr, 2015-06-25:
Which version of cmake and make do you have installed ? I have cmake v3.2.3/make v3.81, using the same compiler as you and the driver compiles.

torleifs, 2015-06-25:
I have cmake 3.0.0 and make 3.81

kenr, 2015-06-26:
You could try to upgrade cmake to the latest version and see if that resolves the issue.

torleifs, 2015-06-26:
Using a full path to the dependencies folder made it compile and link.

kenr, 2015-06-26:
I think we shall keep this one open until the next release and let python or cmake make the SRP path automatically absolute. That way developers will not experience this issue if the documentation is not read thoroughly.

carlescufi, 2016-06-23:
closing after repo overhaul. Reopen if it's still an issue.

closed 2016-06-23 by carlescufi