NordicSemiconductor/pc-ble-driver #2: OSError: ../../../driver/lib/libs130_nrf51_ble_driver.so: wrong ELF class: ELFCLASS64

dinosaurkfb, 2015-06-26:
Hi, I got this error when running the python example.
    `OSError: ../../../driver/lib/libs130_nrf51_ble_driver.so: wrong ELF class: ELFCLASS64`

And it turns out that the library _s130_nrf51_ble_driver.so and libs130_nrf51_ble_driver.so are x86-64 version. Will you supply a 32-bit version?

My environment is:
`Linux dk-ub04 3.13.0-32-generic #57-Ubuntu SMP Tue Jul 15 03:51:12 UTC 2014 i686 i686 i686 GNU/Linux`

kenr, 2015-06-26:
Hi, did you compile the driver from source on your Linux machine, or did you download the precompiled version from the nordicsemi.com pages ?

dinosaurkfb, 2015-06-26:
No I didn't compile it, I just download the driver for Linux from nordicsemi.com.
It required a lot of work to compile it. If you have some precompiled versions, could you give me a download link?

kenr, 2015-06-26:
We do not have any other precompiled versions of the driver, sorry. But it is quite easy to compile the driver. Did you have have any issues when compiling it ?

closed 2015-06-29 by kenr