NordicSemiconductor/pc-ble-driver #4: main.py hangs when calling ble_driver.sd_rpc_open()

dinosaurkfb, 2015-06-29:
Hi, I'm trying the python example of heart_rate_collector and have found out main.py hangs when calling ble_driver.sd_rpc_open() and the same situation occurs when using the example of C code after compiling.
My environment is:
`VirtualBox 4.3.26`
`Linux dk-xub 3.16.0-30-generic #40~14.04.1-Ubuntu SMP Thu Jan 15 17:43:14 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux`
`Python 2.7.6`

The sniffer hardware is a PCA10031 board. It have no problem when using the ble sniffer in Windows 8.1.

So what's the problem might be?

kenr, 2015-06-29:
Have you tried to power-cycle the board before starting the application (not just reset, but power off/on) ? What version of the Segger firmware are you using ?

dinosaurkfb, 2015-06-29:
Yes, I have tried(Plugged in and out.)
I don't know how to find Segger firmware's version. But SEGGER J-Link Commander is V4.92.

kenr, 2015-06-30:
Ok, could you try the following:
Upgrade the Segger firmware to version 4.98e (see https://segger.com/j-link-older-versions.html)

And check that:
1. No other processes are using the serial port file descriptor (/dev/ttyACM*)
2. The user running the BLE driver based application is allowed access to the /dev/ttyACM* file(s)

You could also try to set the log-level to LOG_TRACE with the sd_rpc_log_handler_severity_filter_set or sd_rpc_log_file_severity_filter_set function.

dinosaurkfb, 2015-06-30:
It seems that the COM port's driver doesn't work. I tried 5.04 and 4.98e, but none of them work. With Minicom connecting to a  PCA10028 board, which print log message to UART, I still can't see any log message. Yet this log message can be seen from a Windows version Minicom.
FYI, here is messages from dmesg:
[  433.251733] usb-storage 1-2:1.3: USB Mass Storage device detected
[  433.266244] scsi5 : usb-storage 1-2:1.3
[  455.260975] usb 1-2: reset full-speed USB device number 5 using ohci-pci
[  455.722882] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[ 1284.836195] usb 1-2: USB disconnect, device number 5
[ 1293.453797] usb 1-2: new full-speed USB device number 6 using ohci-pci
[ 1293.899492] usb 1-2: New USB device found, idVendor=1366, idProduct=1015
[ 1293.899495] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1293.899497] usb 1-2: Product: J-Link
[ 1293.899498] usb 1-2: Manufacturer: SEGGER
[ 1293.899499] usb 1-2: SerialNumber: 000681467806
[ 1293.916667] cdc_acm 1-2:1.0: ttyACM0: USB ACM device

PickledGherkin, 2015-11-11:
Did you manage to solve this? I'm getting the same issue on windows 7 with Python and the advertising example. I'm also getting it with  a c# wrapper that I've created as well. The C/C++ advertising example actually works ok.

jonnyhsu, 2015-11-12:
I'm having the same problem with Windows 8.1 and python 2.7. All of the python examples hang on the ble_driver.sd_rpc_open() call, and the C++ advertising example has the same problem.

PickledGherkin, 2015-11-12:
I've managed to solve the problem for me. The com port for the dongle was forced to a specific port because it kept changing. However a TI FET had been installed/connected on the same COM port in the past. When we uninstalled the dongle driver from that com port using USBDeview and allowed the OS to allocate a COM port automatically then the C# and python apps started working. It seemed that this conflict in drivers did not affect the C/C++ example but did affect python and C#.

carlescufi, 2016-06-23:
closing after repo overhaul. Reopen if it's still an issue.

closed 2016-06-23 by carlescufi