NordicSemiconductor/pc-ble-driver #9: SD20.patch doesn't apply to SDK 11

rosek86, 2016-06-15:
Hi,

I'm trying to apply SD20 patch to SDK 11, but it fails on 'ble_gap_evt_auth_status.c', there is no 'ble_gap_evt_auth_status_dec' function in SDK 11. I'm using that patch:

https://github.com/NordicSemiconductor/pc-ble-driver/blob/8a5cb05b459f3f6de9519f8dee06cb407933abf3/hex/SD20.patch

[image]

Can you please suggest a solution?

Many thanks,
Krzysztof

bihanssen, 2016-06-15:
Hi,
the patching is not well documented, sorry about that. The tool you are using is apparently not happy about the format of the patch file.

Try using this on the command line:
patch --ignore-whitespace -p0 -i \repos\pc-ble-driver\hex\SD20.patch

I should also mention that the patch file has been updated as of today with some new fixes. That new patch file should work well with TortoiseGitMerge also.

rosek86, 2016-06-15:
Many thanks, the newest patch applies without any problem. Can you please also confirm what defines should I add to standard Keil project to compile firmware for nRF52-DK?

I'm using Keil project files from 'pca10040\ser_s132_hci\arm5_no_packs'. It contains the following defines:

>  HCI_TIMER2 BLE_STACK_SUPPORT_REQD __STACK_SIZE=2048 __HEAP_SIZE=1024 BOARD_PCA10040 NRF52_PAN_12 NRF52_PAN_15 NRF52_PAN_20 NRF52_PAN_30 NRF52_PAN_31 NRF52_PAN_36 NRF52_PAN_51 NRF52_PAN_53 NRF52_PAN_54 NRF52_PAN_55 NRF52_PAN_58 NRF52_PAN_62 NRF52_PAN_63 NRF52_PAN_64 CONFIG_GPIO_AS_PINRESET S132 SER_CONNECTIVITY APP_SCHEDULER_WITH_PAUSE BSP_DEFINES_ONLY NRF52 SOFTDEVICE_PRESENT SWI_DISABLE0

I think HCI_LINK_CONTROL should be added to these defines.

When I try to compile project I get the errors shown below. Can you suggest what to do with these errors?

> "no source": Error:  #5: cannot open source input file "..\..\..\..\..\..\components\serialization\connectivity\codecs\s130\serializers\ble_gap_evt_conn_param_update_request.c": No such file or directory
> ..\..\..\..\..\..\components\serialization\connectivity\codecs\s130\serializers\ble_gap_evt_conn_param_update_request.c: 0 warnings, 1 error

Should I change anything else in the project configuration? I'm using PCA10040 V1.1.0.

rosek86, 2016-06-15:
Ok, I have these errors because paths to these files were too long, please ignore this part of my question!

rosek86, 2016-06-15:
@bihanssen to confirm, now my compilation works well, I've only added HCI_LINK_CONTROL and APP_SCHEDULER_WITH_PROFILER to project configuration.

Thank you very much for your help!

carlescufi, 2016-06-15:
@rosek86 are you trying out the self_contained_driver branch? It's work in progress. You can also get the precompiled .hex files from the /hex folder

rosek86, 2016-06-15:
@carlescufi I'm using pc-ble-driver @ 8a5cb05 together with pc-ble-driver-js, I've only used firmware patch from self_contained_driver branch. 

I need to add my own modification to the ble driver, the idea is to add Shock Burst mode and send stream data from remote transmitter to the nRF52 receiver. We also we use different pins in our hardware...

bihanssen, 2016-06-15:
@rosek86 HCI_LINK_CONTROL and APP_SCHEDULER_WITH_PROFILER are the extra defines you need to add, that is correct.

rosek86, 2016-06-15:
Thank you, all looks good now.

closed 2016-06-15 by rosek86