NordicSemiconductor/pc-ble-driver #11: driver for raspberry pi

angelovAlex, 2016-06-24:
I tried to compile the driver on raspberry pi. Most of files (87%) were successfully compiled, but there're a lot of compile errors in this file:
`Building CXX object CMakeFiles/pc_ble_driver_obj.dir/src/sd_rpc_impl.cpp.o`

```
In file included from /usr/include/c++/4.9/thread:35:0,
                 from /home/pi/command_line_tools/pc-ble-driver/include/internal/transport/serialization_transport.h:45,
                 from /home/pi/command_line_tools/pc-ble-driver/include/internal/adapter_internal.h:42,
                 from /home/pi/command_line_tools/pc-ble-driver/src/sd_rpc_impl.cpp:40:
/usr/include/c++/4.9/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support is currently experimental, and must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support for the \
  ^
In file included from /home/pi/command_line_tools/pc-ble-driver/include/internal/transport/serialization_transport.h:41:0,
                 from /home/pi/command_line_tools/pc-ble-driver/include/internal/adapter_internal.h:42,
                 from /home/pi/command_line_tools/pc-ble-driver/src/sd_rpc_impl.cpp:40:
/home/pi/command_line_tools/pc-ble-driver/include/internal/transport/transport.h:49:14: error: ‘function’ in namespace ‘std’ does not name a template type
 typedef std::function<void(sd_rpc_app_status_t code, const char *message)> status_cb_t;
              ^
/home/pi/command_line_tools/pc-ble-driver/include/internal/transport/transport.h:50:14: error: ‘function’ in namespace ‘std’ does not name a template type
 typedef std::function<void(uint8_t *data, size_t length)> data_cb_t;
              ^
/home/pi/command_line_tools/pc-ble-driver/include/internal/transport/transport.h:51:14: error: ‘function’ in namespace ‘std’ does not name a template type
 typedef std::function<void(sd_rpc_log_severity_t severity, std::string message)> log_cb_t;
              ^
/home/pi/command_line_tools/pc-ble-driver/include/internal/transport/transport.h:56:27: error: ‘status_cb_t’ has not been declared
     virtual uint32_t open(status_cb_t status_callback, data_cb_t data_callback, log_cb_t log_callback);
                           ^
/home/pi/command_line_tools/pc-ble-driver/include/internal/transport/transport.h:56:56: error: ‘data_cb_t’ has not been declared
     virtual uint32_t open(status_cb_t status_callback, data_cb_t data_callback, log_cb_t log_callback);
                                                        ^
/home/pi/command_line_tools/pc-ble-driver/include/internal/transport/transport.h:56:81: error: ‘log_cb_t’ has not been declared
     virtual uint32_t open(status_cb_t status_callback, data_cb_t data_callback, log_cb_t log_callback);
                                                                                 ^
/home/pi/command_line_tools/pc-ble-driver/include/internal/transport/transport.h:63:5: error: ‘status_cb_t’ does not name a type
     status_cb_t statusCallback;
     ^
/home/pi/command_line_tools/pc-ble-driver/include/internal/transport/transport.h:64:5: error: ‘data_cb_t’ does not name a type
     data_cb_t dataCallback;
```
and lots more...

carlescufi, 2016-06-24:
I see. The RPi has an ARM processor so I haven't really accounted for that or checked it. I will try to build it and report back

carlescufi, 2016-06-24:
I don't have an RPi here now, but this is strange, because on RPi it should default to gcc.cmake which includes -std=c++11 as its compile options. Which CMake version are you using?

angelovAlex, 2016-06-24:
g++:
`g++ (Raspbian 4.9.2-10) 4.9.2`
cmake:
`cmake version 3.0.2`

carlescufi, 2016-06-24:
> cmake version 3.0.2

There's the issue. We require CMake 3.3 or higher. I will push a commit to enforce this, it was mistakenly left out.

carlescufi, 2016-06-24:
Can you report back once you upgrade CMake? Thanks!

angelovAlex, 2016-06-24:
I downloaded cmake 3.3.0 and RPi is still compiling it. I will let you know as soon as it finish compiling.

carlescufi, 2016-06-24:
great! thanks for helping us debug this

angelovAlex, 2016-06-25:
It took ages to compile cmake on RPi, but finally it worked.
The only thing it asked for `libudev`
```
/home/pi/command_line_tools/pc-ble-driver/src/platform/linux/serial_port_enum.cpp:38:21: fatal error: libudev.h: No such file or directory
 #include <libudev.h>
```
Successfully compiled after I did `sudo apt-get install libudev-dev`

Now I'm trying to compile any of examples and I have some issues:
in `nrf51-ble-driver_linux_0.5.0/python` there a file `_s130_nrf51_ble_driver.so`
and all python examples crashes with
`ImportError: ../../_s130_nrf51_ble_driver.so: wrong ELF class: ELFCLASS64`

I also tried to compile `nrf51-ble-driver_linux_0.5.0/driver/examples/heart_rate_collector/gcc`
```
gcc -I../../../include -Wall -c -o main.o ../main.c
gcc -o main main.o -L../../../lib -ls130_nrf51_ble_driver -Wl,-rpath=../../../lib
main.o: In function `main':
main.c:(.text+0x10b8): undefined reference to `sd_rpc_serial_port_name_set'
main.c:(.text+0x10c0): undefined reference to `sd_rpc_serial_baud_rate_set'
main.c:(.text+0x10c8): undefined reference to `sd_rpc_log_handler_set'
main.c:(.text+0x10d0): undefined reference to `sd_rpc_evt_handler_set'
collect2: error: ld returned 1 exit status
../../Makefile.common:46: recipe for target 'main' failed
make: *** [main] Error 1
```

carlescufi, 2016-06-26:
> The only thing it asked for libudev

That's documented in the [pc-ble-driver-py build steps](https://github.com/NordicSemiconductor/pc-ble-driver-py#ubuntu-linux). The thing is that pc-ble-driver is really intended to be built with either the -py or -js bindings.

> Now I'm trying to compile any of examples 

There aren't any examples to compile. All the examples are in pc-ble-driver-py as Python scripts that don't need to be compiled, just run. Not sure what you are doing there.

> I also tried to compile nrf51-ble-driver_linux_0.5.0/driver/examples/heart_rate_collector/gcc

I really am not sure what you are doing here. Where is that example? Are you using pc-ble-driver and pc-ble-driver-py master?

carlescufi, 2016-06-26:
After looking a bit more at your comment it seems like you're using an old version of pc-ble-driver, when we used to require the patched SDK and had C examples. Now, on pc-ble-driver master we only provide the driver itself in the form of a shared library, and pc-ble-driver-py is the repo to use to access that library from Python

carlescufi, 2016-06-26:
If you prefer to use tags or download directly from GitHub instead of cloning with git, I have now tagged the current master as v1.0.0

carlescufi, 2016-06-26:
By the way, the building instructions for this pc-ble-driver repo, as described in the README, can be found [in the pc-ble-driver-py documentation](https://github.com/NordicSemiconductor/pc-ble-driver-py#introduction)

carlescufi, 2016-06-28:
Any news on this?

carlescufi, 2016-06-28:
I just built the whole pc-ble-driver-py, which includes this repo, on:

Linux raspberrypi 4.4.13-v7+ #894 SMP Mon Jun 13 13:13:27 BST 2016 armv7l GNU/Linux

with no issues at all, no warnings.

I will close this issue for now.

angelovAlex, 2016-06-29:
Sorry for the long reply. Unfortunately, I still need some help with it.

Initially I downloaded the driver from https://www.nordicsemi.com/eng/Products/nRF51-Dongle. This page doesn't really say that it contains old version and that new version is available on github.

I tried to compile pc-ble-driver-py and that's what I got:
```
cmake -G "Unix Makefiles" ..
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:20 (include):
  include could not find load file:

    pc-ble-driver/cmake/pc-ble-driver.cmake

CMake Error at CMakeLists.txt:21 (add_subdirectory):
  The source directory

    /home/pi/pc-ble-driver-py/pc-ble-driver/CMakeLists.txt

  does not contain a CMakeLists.txt file.

-- Found SWIG: /usr/bin/swig2.0 (found version "2.0.12") 
-- Found PythonLibs: /usr/lib/arm-linux-gnueabihf/libpython2.7.so (found suitable version "2.7.9", minimum required is "2.7") 
You have called ADD_LIBRARY for library python without any source files. This typically indicates a problem with your CMakeLists.txt file
CMake Error at /usr/local/share/cmake-3.3/Modules/UseSWIG.cmake:298 (message):
  Cannot find Swig library "/usr/lib/arm-linux-gnueabihf/libpython2.7.so".
Call Stack (most recent call first):
  CMakeLists.txt:74 (swig_link_libraries)

CMake Error at CMakeLists.txt:101 (add_custom_command):
  add_custom_command Wrong syntax.  A TARGET or OUTPUT must be specified.

CMake Error at CMakeLists.txt:106 (add_custom_command):
  add_custom_command Wrong syntax.  A TARGET or OUTPUT must be specified.

CMake Error at CMakeLists.txt:111 (add_custom_command):
  add_custom_command Wrong syntax.  A TARGET or OUTPUT must be specified.

CMake Error at CMakeLists.txt:116 (add_custom_command):
  add_custom_command Wrong syntax.  A TARGET or OUTPUT must be specified.

CMake Error at CMakeLists.txt:123 (build_metadata):
  Unknown CMake command "build_metadata".

-- Configuring incomplete, errors occurred!
See also "/home/pi/pc-ble-driver-py/build/CMakeFiles/CMakeOutput.log".
```
Let me know if you need `/home/pi/pc-ble-driver-py/build/CMakeFiles/CMakeOutput.log` it's quite long.

I also tried to install the driver with `pip install pc-ble-driver-py`, and I have runtime error:
```
>>> from pc_ble_driver_py.ble_driver     import *
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pc_ble_driver_py/ble_driver.py", line 94, in <module>
    raise RuntimeError("Could not load shared library {} : '{}'.".format(shlib_path, error))
RuntimeError: Could not load shared library /usr/local/lib/python2.7/dist-packages/pc_ble_driver_py/lib/linux/x86_32/libpc_ble_driver_shared.so : '/usr/local/lib/python2.7/dist-packages/pc_ble_driver_py/lib/linux/x86_32/libpc_ble_driver_shared.so: cannot open shared object file: No such file or directory'.
```

carlescufi, 2016-06-29:
Sorry about the outdated version you got from the link. We will update that, thanks for letting us know.

> I also tried to install the driver with pip install pc-ble-driver-py, and I have runtime error:
Did you do this on the Raspberry Pi?

carlescufi, 2016-06-29:
> I tried to compile pc-ble-driver-py and that's what I got:

You need to clone and then pull the submodules:

```
git clone git@github.com:NordicSemiconductor/pc-ble-driver-py.git
git submodule init .
git submdoule update
```

angelovAlex, 2016-06-29:
> You need to clone and then pull the submodules:

Oh, yes, I forgot to clone submodule, I cloned `pc-ble-driver-py` with `--recursive` option and it is being compiled now.

> Did you do this on the Raspberry Pi?

Yes, it's raspberry pi model B, I believe. 
```
uname -a
Linux raspberrypi 4.4.13+ #894 Mon Jun 13 12:43:26 BST 2016 armv6l GNU/Linux
```

carlescufi, 2016-06-29:
I compiled cmake and swig from source on the Raspberry Pi. Didn't take long but then againi I have a RPi 3, which is quite fast. I can give you a compiled version of the library for Pi if you want in the meantime, since we don't "officially" support it.

carlescufi, 2016-06-29:
>> Did you do this on the Raspberry Pi?
> Yes, it's raspberry pi model B, I believe.

Then it's normal. The pc-ble-driver-py package in pypi doesn't contain a shared library compiled for the Raspberry Pi

carlescufi, 2016-06-29:
Here's a compiled version for Raspberry Pi in case you run into issues:

[pc_ble_driver_py_rpi-0.5.0.tar.gz](https://github.com/NordicSemiconductor/pc-ble-driver/files/339470/pc_ble_driver_py_rpi-0.5.0.tar.gz)

angelovAlex, 2016-06-29:
It is compiled successfully. And thanks for sharing your compiled version of the library.
Now can you point me on how to run any of examples that are provided with the library?
I ran `sudo python setup.py install` and after installing, tried to run an example:
```
python heart_rate_collector.py 
	0 : /dev/ttyACM0: 000480107551
 
Enter your choice: 0
Serial port used: /dev/ttyACM0
ERROR:pc_ble_driver_py.ble_driver:Unable to find serial port
Traceback (most recent call last):
  File "heart_rate_collector.py", line 155, in <module>
    main(serial_port)
  File "heart_rate_collector.py", line 119, in main
    driver    = BLEDriver(serial_port=serial_port, auto_flash=True)
  File "/usr/local/lib/python2.7/dist-packages/pc_ble_driver_py-0.5.0-py2.7.egg/pc_ble_driver_py/ble_driver.py", line 772, in __init__
    flasher = Flasher(serial_port=serial_port)
  File "/usr/local/lib/python2.7/dist-packages/pc_ble_driver_py-0.5.0-py2.7.egg/pc_ble_driver_py/ble_driver.py", line 691, in __init__
    raise NordicSemiException('nrfjprog not installed')
pc_ble_driver_py.exceptions.NordicSemiException: nrfjprog not installed
```
so where I can get nrfjprog for RPi?

carlescufi, 2016-06-29:
> I ran sudo python setup.py install and after installing, tried to run an example:

Change `auto_flash` to False so that the driver doesn't try to flash the firmware:
https://github.com/NordicSemiconductor/pc-ble-driver-py/blob/master/python/pc_ble_driver_py/examples/heart_rate_collector.py#L119

> so where I can get nrfjprog for RPi?
There isn't one. You will  need to flash your Devkit on a computer, see here:

https://github.com/NordicSemiconductor/pc-ble-driver/tree/self_contained_driver#hardware-setup

carlescufi, 2016-06-29:
I recommend trying the advertising example first, it's simpler and an easy way to check if your setup works

angelovAlex, 2016-06-29:
I hope I'm not bothering you :)
```
python advertising.py /dev/ttyACM0
Serial port used: /dev/ttyACM0
Traceback (most recent call last):
  File "advertising.py", line 74, in <module>
    main(sys.argv[1])
  File "advertising.py", line 63, in main
    driver.open()
  File "/usr/local/lib/python2.7/dist-packages/wrapt/wrappers.py", line 561, in __call__
    args, kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pc_ble_driver_py-0.5.0-py2.7.egg/pc_ble_driver_py/ble_driver.py", line 112, in wrapper
    raise NordicSemiException('Failed to {}. Error code: {}'.format(wrapped.__name__, err_code))
pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: 13
```

carlescufi, 2016-06-29:
>pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: 13

Try giving your user permissions to access the serial port:
https://github.com/NordicSemiconductor/pc-ble-driver/tree/master#ubuntu-linux

angelovAlex, 2016-06-29:
Tried. Still the same. Also tried with `sudo`.

carlescufi, 2016-06-29:
hmm strange. And you are sure that your board is flashed with the connectivity firmware as per the instructions here:

https://github.com/NordicSemiconductor/pc-ble-driver/tree/master#flashing-the-connectivity-firmware

carlescufi, 2016-06-29:
Oh by the way, are you using the dongle or the nRF51 or nRF52 DK? I am not sure the connectivity firmware is compatible with the dongle, it definitely works fine with the DK

angelovAlex, 2016-06-29:
yea, it's definatly flashed using nrfgo studio. 
I was working on Windows, I ran heart_rate_collector.py  and it was showing me beacons of other devices.
I tried to reprogram it with firmware from `hex` folder from your archive, it didnt help.

carlescufi, 2016-06-29:
Error 13 is timeout, so the something is up with the communication over the serial port. Can you try updating the segger j-link drivers to the latest on you Pi?

angelovAlex, 2016-06-29:
Not sure how to do it.

carlescufi, 2016-06-29:
You need this file: 
https://www.segger.com/downloads/jlink/JLink_Linux_V512g_arm.tgz

From here:
https://www.segger.com/downloads/jlink

angelovAlex, 2016-06-29:
not sure how to install it propertly. I cp 99-rules file and rebooted rasberry pi. Should I copy anything else from that archive?

carlescufi, 2016-06-29:
I don't think so, that's all I did myself. I honestly am out of ideas. Have you tried the version that I myself compiled?

carlescufi, 2016-06-29:
Wait, which connectivity firmware .hex are you flashing? You should be flashing this one:

https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/hex/connectivity_115k2_with_s130_2.0.1.hex

You need at least a dongle with an nRF51 chip that supports the s130_nrf51_2.0.0 SoftDevice and SDK 11.0. You can check compatibility here:

https://www.nordicsemi.com/eng/nordic/download_resource/41917/8/16739800

angelovAlex, 2016-06-29:
yeah, I was using this firmware for linux drive.
For Windows old drive it works with `s130_1.0.0`.
According to the table 3 and 4 I have IC rev. 2 that doesnt support `S130`, but it works on Windows perfectly. The chip on it is `nrf51822 QFAAG0`

carlescufi, 2016-06-29:
Let me get this straight: On Windows have you tried the latest (current) version by running `pip install pc-ble-driver-py` and it works, is that correct?

carlescufi, 2016-06-29:
Because the first thing to do is to run this exact version, installed from pip (and not downloaded from nordicsemi.com), on Windows (or Linux or OS X) to make sure your hardware supports the current connecitivity firmware based on SoftDevice S130 2.0.0

angelovAlex, 2016-06-29:
No, on Windows I tried old version that's on nordic's website with s130_1.0.0 firmware. Ok, I will try the latest firmware now.

carlescufi, 2016-06-29:
Ah then it could be that the new firmware doesn't support your hardware, which is revision 2. Unfortunately the old firmware is not compatible with the new pc-ble-driver-py, and the old pc-ble-driver from the web is not buildable or runnable on the raspberry pi :(

angelovAlex, 2016-06-29:
yeah, I have the same error on Windows now. So s130 v1.0.0 is compatible, but v2.0.0 is not. That's a bit strange as I thought that was supposed to be the same softdevice but with small changes and bug fixes. I have PCA10003 seems to have revision 3, I will try with this board now.

angelovAlex, 2016-06-29:
nrfjprog doesn't want to program PCA10003 anymore. I should have more dongles somewhere, will try to find them tomorrow.

carlescufi, 2016-07-01:
Did you try it with more recent hardware?

carlescufi, 2016-07-05:
@angelovAlex, did it end up working for you?

angelovAlex, 2016-07-06:
no, I haven't found any new nordic's dongle. I have thousands of boards with nrf51422 and my goal was to use one of them as ble device for raspberry pi. Unfortunately, it can't be used directly without segger, so only usb dongles or nrf51422 + segger. I could try it with segger, but it's too complicated solution for me. So I decided to just buy a ble dongle that is compatible with raspberry pi.
Thank you for your help.

carlescufi, 2016-07-07:
No problem, closing this issue now.

closed 2016-07-07 by carlescufi

gilarus, 2016-12-13:
i have the same problem! 

`Serial port used: /dev/ttyACM0
Traceback (most recent call last):
  File "advertising.py", line 74, in <module>
    main(sys.argv[1])
  File "advertising.py", line 63, in main
    driver.open()
  File "/usr/local/lib/python2.7/dist-packages/wrapt/wrappers.py", line 561, in __call__
    args, kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pc_ble_driver_py-0.5.0-py2.7.egg/pc_ble_driver_py/ble_driver.py", line 112, in wrapper
    raise NordicSemiException('Failed to {}. Error code: {}'.format(wrapped.__name__, err_code))
pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: 13`
how to sovle it ?

tarunsmalviya, 2018-09-25:
> Here's a compiled version for Raspberry Pi in case you run into issues:
> 
> [pc_ble_driver_py_rpi-0.5.0.tar.gz](https://github.com/NordicSemiconductor/pc-ble-driver/files/339470/pc_ble_driver_py_rpi-0.5.0.tar.gz)

I tried after installing pc_ble_driver_py_rpi and I am getting this error. Can you please help me.
[image]

hallmatthewp, 2019-02-05:
I'm running into the same issue as @tarunsmalviya, above, on my raspi.

tarunsmalviya, 2019-02-06:
@hallmatthewp You need to compile pc_ble_driver_py for arm.

1. Firstly remove the nrfutil and freshly install it again.

2. Follow [this](https://github.com/NordicSemiconductor/pc-ble-driver-py/issues/57#issue-341766303) to generate pc_ble_driver_py binaries.

3. Then copy all files from **pc-ble-driver-py/build/outdir** to **/usr/local/lib/python2.7/dist-packages/pc_ble_driver_py/lib/linux/x86_32**