NordicSemiconductor/pc-ble-driver #12: compile problem on windows

Beeeep, 2016-07-01:
I tried to compile the driver on windows by command "python ..\pc-ble-driver-master\scripts\build.py -srp E:\WorkSetting\pc-ble-driver-old\boost_1_56_0 -d -b". Most of files (81%) were successfully compiled, but there're a lot of compile errors in this file:

[ 81%] Building CXX object driver/CMakeFiles/s130_nrf51_ble_driver.dir/E_/WorkSe
tting/pc-ble-driver-old/boost_1_56_0/nRF51_SDK_8.1.0_b6ed55f/components/serializ
ation/common/struct_ser/s130/ble_gap_struct_serialization.c.obj
E:\WorkSetting\pc-ble-driver-old\boost_1_56_0\nRF51_SDK_8.1.0_b6ed55f\components
\serialization\common\struct_ser\s130\ble_gap_struct_serialization.c: In functio
n 'uint32_t ble_gap_opt_scan_req_report_t_dec(const void*, uint32_t, uint32_t*,
void*)':
E:\WorkSetting\pc-ble-driver-old\boost_1_56_0\nRF51_SDK_8.1.0_b6ed55f\components
\serialization\common\struct_ser\s130\ble_gap_struct_serialization.c:1275:58: er
ror: invalid conversion from 'const void*' to 'const uint8_t* {aka const unsigne
d char*}' [-fpermissive]
     err_code = uint8_t_dec(p_buf, buf_len, p_index, &byte);
                                                          ^
In file included from E:\WorkSetting\pc-ble-driver-old\boost_1_56_0\nRF51_SDK_8.
1.0_b6ed55f\components\serialization\common\struct_ser\s130\ble_gap_struct_seria
lization.c:14:0:
E:/WorkSetting/pc-ble-driver-old/boost_1_56_0/nRF51_SDK_8.1.0_b6ed55f/components
/serialization/common/ble_serialization.h:221:10: note: initializing argument 1
of 'uint32_t uint8_t_dec(const uint8_t*, uint32_t, uint32_t*, void*)'
 uint32_t uint8_t_dec(uint8_t const * const p_buf,
          ^
driver\CMakeFiles\s130_nrf51_ble_driver.dir\build.make:3312: recipe for target '
driver/CMakeFiles/s130_nrf51_ble_driver.dir/E_/WorkSetting/pc-ble-driver-old/boo
st_1_56_0/nRF51_SDK_8.1.0_b6ed55f/components/serialization/common/struct_ser/s13
0/ble_gap_struct_serialization.c.obj' failed
mingw32-make[2]: *** [driver/CMakeFiles/s130_nrf51_ble_driver.dir/E_/WorkSetting
/pc-ble-driver-old/boost_1_56_0/nRF51_SDK_8.1.0_b6ed55f/components/serialization
/common/struct_ser/s130/ble_gap_struct_serialization.c.obj] Error 1
CMakeFiles\Makefile2:86: recipe for target 'driver/CMakeFiles/s130_nrf51_ble_dri
ver.dir/all' failed
mingw32-make[1]: *** [driver/CMakeFiles/s130_nrf51_ble_driver.dir/all] Error 2
Makefile:82: recipe for target 'all' failed
mingw32-make: *** [all] Error 2
CRITICAL:serialization_dll:Failed to build artifact driver. Error code: driver.
INFO:..\pc-ble-driver-master\scripts\build.py:Failed to build artifact driver. Error code: driver.

My compile environment:
gcc 4.9.3
cmake 3.6.0-rc2
boost 1.56.0
python 2.7.11
swig 2.0.12

bihanssen, 2016-07-01:
Hi,
I see you are using the old version of pc-ble-driver (tag 0.5?). If you can move to using [pc-ble-driver-py](https://github.com/NordicSemiconductor/pc-ble-driver-py) instead then I would strongly suggest so as that is the solution we will be using forward. pc-ble-driver-py is the python binding part of pc-ble-driver split out into a separate repository. There has also been major rewrites and improvements to the repos.

If you still wish to go with the old pc-ble-driver and try to resolve the build issue then you could try to downgrade to python 2.7.6 according to issue #7.

Beeeep, 2016-07-04:
I'm using  pc-ble-driver (tag 0.5).  After downgrade to python 2.7.6,it's still  the same compile error.

bihanssen, 2016-07-04:
Try also downgrading mingw/gcc to the version that is linked to, version 4.8.2. I believe 4.9 will not be compatible.

bihanssen, 2016-07-07:
Hi, did it work out downgrading to gcc 4.8.2?

bihanssen, 2016-09-21:
Closing the issue due to inactivity. Please reopen if this is still a problem.

closed 2016-09-21 by bihanssen