NordicSemiconductor/pc-ble-driver #13: Compiling the nRF5x connectivity firmware for a custom board

metc, 2016-07-06:
Hi,

I am very interested in this project. I will probably getting started using the compiled hex files you provide for the nRF51 or 52 DK, but I would like to have some information on how they are generated. 

Could you please provide a bit more information on how both `connectivity_115k2_with_s130_2.0.1.hex` and `connectivity_115k2_with_s132_2.0.1.hex` hex files are generated ? Are they generated from the SDK11  `ble_connectivity` example and then merge with the SoftDevice ? Are any other SDK compatible too ?

The lack of documentation is a bit a problem for me because I would like to compile the ble connectivity project for a custom board. On the embedded side, I need at least to modify the UART settings and pins.

I can see many targets available with the `ble_central_and_peripheral\ble_connectivity\` example from the SDK11 (ble_connectivity_`SoftDevice` _ `transport` _ `board`). What is the difference between the UART and HCI versions ? Are they both compatible with this PC library/driver ? The HCI commands are proprietary commands ?

@bihanssen In the issue #9, you give some information about the SDK patch and the flags we should use to compile the project. It's quite hard to read the issues to be able to compile the connectivity application. I think a dedicated page on the project Wiki could help.

This is more a discussion than an issue, sorry for that. Any help/explanations are welcomed.

bihanssen, 2016-07-07:
Hi, 
here is the recipe for generating connectivity firmware hex file:

* Download and unpack `nRF5_SDK_11.0.0_89a8197.zip`
* Apply patch SD20_SDK11.patch from unpacked SDK folder (e.g. `git apply -p1 --ignore-whitespace /repos/pc-ble-driver/hex/SD20_SDK11.patch`)
* Open /examples/ble_central_and_peripheral/ble_connectivity/pca100XX/ser_s13X_hci project in the compiler of choice
* Add defines APP_SCHEDULER_WITH_PROFILER and HCI_LINK_CONTROL
* Compile
* Merge the built connectivity hex file with corresponding SoftDevice hex file (e.g. `mergehex -m connecitivy.hex softdevice.hex -o connectivity_with_softdevice.hex`)

carlescufi, 2016-07-07:
This is now part of the official doc:
https://github.com/NordicSemiconductor/pc-ble-driver#compiling-the-connectivity-hex-files

metc, 2016-07-07:
Hi @bihanssen and @carlescufi 

Thanks for the information and the updated documentation :notes: 

The `pc-ble-driver` communicates over serial with the `ser_s13X_hci` connectivity firmware. What is the difference between the `ser_s13X_uart` and the `ser_s13X_hci` versions ? I am not sure to understand what is different because th transport layer is the same ?

bihanssen, 2016-07-07:
The `hci` variant uses the Bluetooth defined _Three-wire UART Transport Layer_. This includes error protection, handshaking, SLIP enoding, etc. The `uart` variant is simpler and does not add any protection against transport errors. To be compatible with pc-ble-driver you must use the `hci` variant.

metc, 2016-07-07:
Thank you, it's more clear now. I also find detailed information [on the infocenter](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v11.0.0%2Flib_hci_transport.html&cp=6_0_0_3_28_0) about the supported feature of the HCI transport library you provide.

Last question: are the HCI commands vendor specific ? For instance, is [the start scan HCI command](https://github.com/NordicSemiconductor/pc-ble-driver/blob/8f4fbc88ab16a1384ba5069647ac2691e17a760e/src/sdk/codec/ble_gap_scan_start.c#L50) a proprietary command ?

carlescufi, 2016-07-07:
Yes, they are vendor specific. In fact they reflect the SoftDevice API. We don't have an HCI implementation of the Link Layer, that is, we don't have a standard HCI interface to the chip.

metc, 2016-07-08:
I am now able to debug the embedded connectivity application. Thanks for the detailed answers and the updated documentation.

closed 2016-07-08 by metc