NordicSemiconductor/pc-ble-driver #14: [This is not an issue]What does "adapter_t" mean?

HeXiang708, 2016-07-08:
Hello guys,

I find a new parameter "adapter_t" in every function, I want to know what that mean?

Thanks in advance for your help.

Xiang

CoqRogue, 2016-07-08:
One of the things with the updated api for the connectivity project is that we wanted multiadaptr support. `adapter_t` gives us that. When you call `sd_rpc_adapter_create`, found in `sd_rpc.h`, you get an `adapter_t` returned. Sending this in to all the other functions will allow you to differentiate between the adapters you are connected with. Our [javascript bindings](https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/master/src/driver.cpp#L649) may give you an idea on how to use it.

HeXiang708, 2016-07-11:
Thank you! Your javascript bindings help me so much.

closed 2016-07-11 by HeXiang708