NordicSemiconductor/pc-ble-driver #15: SDK path for hex breaks gcc buid

davidqmora, 2016-07-18:
The recent instructions for building the hex say to apply a path from the repo to the SDK. That seems to work well; however, gcc does not like the line:

static const version_info_t version_info __attribute__((at(0x20000))) = {0x46D8A517, 1, 0xFFFFFF, 0x12345678, 1, 0, 0, 0, NULL};

As it does not recognize at(). That looks like a Keil artifact.

main.c:47:1: error: 'at' attribute directive ignored [-Werror=attributes]
 static const version_info_t version_info __attribute__((at(0x20000))) = {0x46D8A517, 1, 0xFFFFFF, 0x12345678, 1, 0, 0, 0, NULL}

 Any suggestions on ho to fix this would be appreciated.

metc, 2016-07-18:
I also try to build the connectivity application using `gcc`, but the provided patch is Keil specific. I think this is still a WIP and I hope a patch will be provided soon (or a new compatible SDK version :notes:).

I had the same issues as you, but I did not had the time to make a PR yet.
First, you can use the following defines for the packed structure:

```c
#if defined(__CC_ARM)
    #define PACKED(TYPE) __packed TYPE
#elif defined(__ICCARM__)
    #define PACKED(TYPE) __packed TYPE
#else
    #define PACKED(TYPE) TYPE __attribute__((packed))
#endif
```

Then, for the `at` attribute, more changes are required (see [this post](http://stackoverflow.com/questions/4067811/how-to-place-a-variable-at-a-given-absolute-address-in-memory-with-gcc)). If you want to use this with `gcc`, you have to add a section attribute in the linker script and then use something like this (have a look at the dfu bootloader example):

```c
#if defined ( __CC_ARM )
  static const version_info_t version_info __attribute__((at(0x20000))) = {0x46D8A517, 1, 0xFFFFFF, 0x12345678, 1, 0, 0, 0, NULL};
#elif defined (__GNUC__ )
  static const version_info_t version_info __attribute__((section(".mySection"))) = {0x46D8A517, 1, 0xFFFFFF, 0x12345678, 1, 0, 0, 0, NULL};
#elif defined (__ICCARM__ )
// TODO
#endif
```

As a workaround for GCC, you could add something like this in the `main` function:

```c
version_info_t *ptr = (version_info_t*)0x20000;
memcp(...); // TODO
```

Nordic, I will also be interested in having a generic patch (or at least compatible with GCC) because I would like to compile the connectivity application for a custom board.

bihanssen, 2016-07-19:
Hi, it is correct that the patch is Keil specific, this is simply because that was what we needed initially. I'll add it to the backlog to create a more generic path. If anyone else wants to take on this task we would gladly accept a PR.

metc, 2016-10-12:
@davidqmora If you want to build the connectivity application with the current patch, as a simple workaround you could comment the line 47 where the `version_info` is declared.

You will be able to compile the application with gcc. Then, when you connect the board with nRF Connect, you will see a message like this because no version is defined:
[image]

Just press "Cancel" in order to use the connectivity application you just compiled. It works great in my case.

bihanssen, 2017-03-21:
The struct in question is currently wrapped in `#if defined ( __CC_ARM )`, so at least it doesn't break when using other compilers. 
Closing the issue.

closed 2017-03-21 by bihanssen