NordicSemiconductor/pc-ble-driver #17: Port of examples to V1.0.0 missing

SwissKnife, 2016-09-30:
The examples are no more usable with PC BLE Driver 1.0.0

Please release the examples for this version.

A "How to.." document would be fine too

bihanssen, 2016-10-03:
Hi, thanks for the feedback. 
Adding examples are in the backlog. Keeping this issue open as a reminder.

SwissKnife, 2016-10-03:
Here is a first guess how to use the Version 1.0.0: It receives advertising packages, but returns error when trying to connect...
[main.zip](https://github.com/NordicSemiconductor/pc-ble-driver/files/506282/main.zip)

bihanssen, 2016-10-05:
Hi,
I tried running your example and it returns error code 18 (NRF_ERROR_CONN_COUNT) when trying to connect. You need to add params for connection count to your ble_enable_params, like this:

```
ble_enable_params.gap_enable_params.periph_conn_count = 1;
ble_enable_params.gap_enable_params.central_conn_count = 7;

```

SwissKnife, 2016-10-06:
Thanks for feedback. 
Currently I use the following settings to connect to one peripheral device. With higher numbers I got some other error (NO_MEM may be), but I have not yet investigated...
```
ble_enable_params.gap_enable_params.periph_conn_count = 0; 
ble_enable_params.gap_enable_params.central_conn_count = 1;
ble_enable_params.gap_enable_params.central_sec_count = 0;   // number of security manager protocol (SMP) instances
```

bihanssen, 2016-11-18:
Hi, various combinations of enable_params will exhaust the available memory. I'm closing this issue now, please reopen if you have follow up questions.

closed 2016-11-18 by bihanssen