NordicSemiconductor/pc-ble-driver #19: Custom/proprietary commands over serial

metc, 2016-10-21:
*This is more a feature request/discussion than an issue.*

Hi,

I am developing a test application on the PC for a custom board running the connectivity application on an nRF51. I made a basic advertising application running on the PC side to control the nRF51 device on my custom board.

Now, I would like to use serial communication to transport custom data packets (other than the `SD_` functions) to control a GPIO or any other peripheral (ADC, PWM) on the nRF SoC from the PC side, using a custom protocol. I think this is not a common use case and the SDK must be modified quite a lot...  What would be the best way to do this ?

Is it be possible to add specific opcodes and custom encoder/decoder functions to be able to use custom data packets on the serial line ? IMO this require quite a lot of modifications in the SDK...

Is it possible to share the serial communication to transport other data than the SD serialized data or should I introduce a new type of packet ? Sequence diagrams and protocol frames could help to better understand the serial communication.

bihanssen, 2016-11-18:
Hi,
I'm not sure what would be the best way to do it, but one suggestion could be to provide your own encoder/decoder functions following the same pattern as for the existing softdevice commands and events, making sure to add the matching functions on both sides (ble-driver and connectivity). It is possible to use the transport layer for other types of packets, as long as the protocol is followed.

If you haven't already you could have a look at the SDK documentation for Serialization: http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v12.0.0/lib_serialization.html?resultof=%22%73%65%72%69%61%6c%69%7a%61%74%69%6f%6e%22%20%22%73%65%72%69%61%6c%22%20

I'll leave this issue open for a while in case anyone else want to pitch in.

bihanssen, 2017-03-21:
Closing the issue due to inactivity. If the issue is still relevant, please try posting on [devzone](https://devzone.nordicsemi.com/questions/).

closed 2017-03-21 by bihanssen