NordicSemiconductor/pc-ble-driver #21: Failed to enable BLE stack when turn on the ATT_MTU config

rick1082, 2016-11-14:
Hello

I tried to run the heart_rate_collector example with "connectivity_1.0.1_115k2_with_s132_3.0.hex".
But the program crashed when doing the "sd_ble_enable()".
The error code is 0x04 which indicates that the memory assigned to Softdevice is not enough. 
Without enabled the “ATT MTU config”, the example project can run correctly.

Thanks for your attention.

carlescufi, 2016-11-14:
What IC are you using, nRF51 or nRF52?

rick1082, 2016-11-14:
I'm using nRF52(nRF52-DK, PCA10040), thank you :)

bihanssen, 2016-11-15:
@rick1082 The values in the example were not valid due to insufficient memory. I have amended the example with new values so it should be ok now.

rick1082, 2016-11-15:
Thank you @bihanssen, seems the issue can be closed. There's no error when enabled the ATT_MTU config.

closed 2016-11-15 by rick1082