NordicSemiconductor/pc-ble-driver #23: sd_ble_gattc_characteristics_discover() has different result between vendor specific UUID and bluetooth SIG UUID

rick1082, 2016-11-17:
Hello,

I was trying to modify the heart rate collector example to interact with ble_app_uart project.
But I found that when calling sd_ble_gattc_characteristics_discover() to do the characteristic discovery for NUS, I only can get a characteristic in each time in the result. 
For heart rate collector example which interacting with bluetooth SIG profile, the response after calling sd_ble_gattc_characteristics_discover() can be an array which stored all the characteristic which are found, not only the single one. 

Thanks for your attention :)

rick1082, 2016-11-18:
And would the pc-ble-driver add the ble_db_discovery module for helping the service discovery ?

bihanssen, 2016-11-18:
Hi @rick1082,
it is correct that when discovering custom (128 bit) UUIDs you will only be able to discover one UUID for each call to sd_ble_gattc_characteristics_discover(). This is because that is all that can fit in one ATT packet when the ATT_MTU is the default of 23.

Regarding adding ble_db_discovery module, there is no current plan to do so. We do however provide higher level apis in the pc-ble-driver-js and pc-ble-driver-py repos.

rick1082, 2016-11-21:
Hello @bihanssen ,
Thanks for your information, it very clear :)

closed 2016-11-21 by rick1082