NordicSemiconductor/pc-ble-driver #25: How to play with 1Mbps version hex

volca, 2016-11-30:
I tried the 115200 version and it works fine. So I tried another 1Mbps version hex connectivity_1.0.1_1m_with_s132_3.0.hex but not lucky.

I tested it with pc-ble-driver-py.  I changed a little code for heart_rate example. 

```
    driver    = BLEDriver(serial_port=serial_port, baud_rate=1000000, auto_flash=False)
```

 How to change the baud rate to 1Mbps?

volca, 2016-11-30:
More details:

* OS is macOS
* pcb10040

Here's the error message

```
Serial port used: /dev/tty.usbserial-A903J5TX
Traceback (most recent call last):
  File "scanner.py", line 181, in <module>
    main(serial_port)
  File "scanner.py", line 140, in main
    collector.open()
  File "scanner.py", line 67, in open
    self.adapter.driver.open()
  File "/Users/xxxxxx/Library/Python/2.7/lib/python/site-packages/pc_ble_driver_py/ble_driver.py", line 126, in wrapper
    raise NordicSemiException('Failed to {}. Error code: {}'.format(wrapped.__name__, err_code))
pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: 13
```

bihanssen, 2016-12-02:
Hi, it appeared that the hex file for v3 1M was not correctly built. Please try the newly commited hex.

volca, 2016-12-03:
I tried again today and got new error message

```
#python scanner.py nrf52 /dev/tty.usbmodemFD1211 
Serial port used: /dev/tty.usbmodemFD1211
libc++abi.dylib: terminating with uncaught exception of type boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::system::system_error> >: set_option: Invalid argument
Abort trap: 6
```

How can I change the baud rate to 1000000?

bihanssen, 2016-12-05:
Hi, I believe the issue here is with MacOS itself. For some reason it seems not possible to set a port baud rate as high as 1M on macOS. It does work on Linux and Windows.

volca, 2016-12-06:
I tested with another Raspberry PI with ubuntu and it works fine. Thank you very much.

closed 2016-12-06 by volca