NordicSemiconductor/pc-ble-driver #26: start packet of ble connectivity and pc-end

gilarus, 2016-12-14:
hello,
1,  i was try to communicate with my connectivity board with ubuntu15.10 64 desktop.
and now i face a problem, i compiled [pc-ble-driver examples](https://github.com/NordicSemiconductor/pc-ble-driver/tree/master/examples/heart_rate_collector) and pc-ble-driver-py binding example name advertising.py[advertising.py](https://github.com/NordicSemiconductor/pc-ble-driver-py/blob/master/python/pc_ble_driver_py/examples/advertising.py) , and run these two , unfortunately, they caused the same err. err code is 0x0D(13), which means TIMEOUT.

2,  And i try to dump what these two examples send out.

these two examples send out the data below:

`
c0 00 05 00 fb c0 c0 00 2f 00 d1 01 7e c0 .....
`

what does it mean?

after my ubuntu send out the data through /dev/ttyUSB0, the connectivity board has not any response, and it cause TIMOUT(0x0d).
i'm puzzle.
please please help me.

bihanssen, 2016-12-14:
Make sure you have the correct connectivity firmware on the board. Try flash programming connectivity firmware onto the board with nrfjprog.

gilarus, 2016-12-14:
i'm sure the connectivity firmware is correct.
i have found something , in hci_slip_event_handler function, there are three **if** statements (branch):

`
if (packet_type == PKT_TYPE_ACK )
else if ( packet_type == PKT_TYPE_VENDOR_SPECIFIC )
else if (packet_type == PKT_TYPE_LINK_CONTROL)
`

and the value of these three MACROs:

`
PKT_TYPE_ACK -> 0
PKT_TYPE_VENDOR_SPECIFIC -> 14
PKT_TYPE_LINK_CONTROL -> 15
`

and there has not 0x0**5**, which the example programs send out!

`
c0 00 **05** 00 fb c0 c0 00 2f 00 d1 01 7e c0 .....
`

is there something wrong?

bihanssen, 2016-12-14:
You need to look at the connectivity source code after having applied the required patch [descrided in the README](https://github.com/NordicSemiconductor/pc-ble-driver#compiling-the-connectivity-hex-files). Then you will get handling of packet type 5, RESET packet.

gilarus, 2016-12-14:
@bihanssen Thank you, 
it's my stupid, i forgot to patch in new respo. After I patched [sdk110_connectivity.patch](https://github.com/NordicSemiconductor/pc-ble-driver#compiling-the-connectivity-hex-files) , it works.
sorry for that.

closed 2016-12-14 by gilarus