NordicSemiconductor/pc-ble-driver #38: Link pc-ble-driver and the python to use on a Raspberry PI

budius, 2017-04-11:
My issues are related to https://github.com/NordicSemiconductor/pc-ble-driver/issues/11
But that's a fairly long and closed issue, I'm just linking and gonna post my status here, hoping for some help:

hardware setup
- using a nRF51 DK
- flashed the latest connectivity hex (connectivity_1.1.0_115k2_with_s130_2.0.1) using nRFgo Studio on windows (https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/hex/sd_api_v2/connectivity_1.1.0_115k2_with_s130_2.0.1.hex)

software setup, executed all the following steps both on my local x86_64 Linux computer and on a raspberry PI 3 (for test and comparison purposes)
- cloned/compilation of boost and pc-ble-driver
- cloned pc-ble-driver-py
- `pip install pc-ble-driver-py`
- executed `python setup.py install` from pc-ble-driver-py/python folder
- changed the heart rate example to `auto_flash=False`
- executed the heart rate example `sudo python heart_rate_collector.py NRF51`

results:
- on my local x86_64 linux computer I started getting advertisements from Bluetooth devices in the area
- on the raspberry PI 3 it failed to locate the pc_ble_driver (it points to a x86 version)

```
RuntimeError: Failed to locate the pc_ble_driver shared library: /usr/local/lib/python2.7/dist-packages/pc_ble_driver_py-0.8.0-py2.7.egg/pc_ble_driver_py/lib/linux/x86_32/libpc_ble_driver_shared_sd_api_v2.so.
```

so then on the PI I got the pre-compiled supplied here (https://github.com/NordicSemiconductor/pc-ble-driver/issues/11#issuecomment-229373516) and I got the following error:

```
Serial port used: NRF51
Traceback (most recent call last):
  File "heart_rate_collector.py", line 155, in <module>
    main(serial_port)
  File "heart_rate_collector.py", line 122, in main
    collector.open()
  File "heart_rate_collector.py", line 61, in open
    self.adapter.driver.open()
  File "/usr/local/lib/python2.7/dist-packages/pc_ble_driver_py-0.5.0-py2.7.egg/pc_ble_driver_py/ble_driver.py", line 112, in wrapper
    raise NordicSemiException('Failed to {}. Error code: {}'.format(wrapped.__name__, err_code))
pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: 3

```

and I'm completely stuck on it. Running as sudo doesn't help.
We got a project with a few hundreds of the nRF51822 on a WT51822 module and it would be great to get some support on it from Nordic. Also I'm pretty sure that there're other devs out there that would appreciate a smoother ARM integration.

bihanssen, 2017-04-12:
Hi, I haven't got a Raspberry Pi here so can't test it directly. Since the pip release doesn't contain precompiled versions for RPi the best option would be to build from source locally on the RPi. I can try to guide you through that, but basically it involves following the steps found in the readme guide [compiling from source](https://github.com/NordicSemiconductor/pc-ble-driver-py#compiling-from-source).
You could also try posting on devzone in case others have similar experiences.

falkorichter, 2017-04-12:
@budius cross post to https://devzone.nordicsemi.com/questions/ sounds like a good idea.

carlescufi, 2017-04-12:
@budius I am the original author of that RPi build. What you can do is, after you've built pc-ble-driver you'll get a set of shared objects (.so) in the build folder. Copy those to:

`/usr/local/lib/python2.7/dist-packages/pc_ble_driver_py-0.8.0-py2.7.egg/pc_ble_driver_py/lib/linux/x86_32/`

and hopefully that should fix it.

carlescufi, 2017-04-12:
Actually, what are you trying to achieve exactly? I just built pc-ble-driver-py from source on my RPi 3 to check that it works, and I had no problem. Here's my command-line history:

```
sudo apt install cmake
sudo apt install swig
sudo apt install git make gcc g++
sudo apt install cmake swig libudev-dev python python-dev
cd prog/
unzip ../boost_1_63_0.zip
export BOOST_ROOT=~/prog/boost_1_63_0/
cd $BOOST_ROOT
./bootstrap.sh
./b2 toolset=gcc cxxflags=-fPIC cflags=-fPIC address-model=32 link=static --with-thread --with-system --with-regex --with-date_time --with-chrono
cd ~/src
git clone git@github.com:NordicSemiconductor/pc-ble-driver-py.git
cd pc-ble-driver-py/
git submodule init
git submodule update
cd build/
cmake -G "Unix Makefiles" ..
make
cd ..
cd python/
cp pc_ble_driver_py/examples/advertising.py .
python advertising.py
python advertising.py NRF51
python advertising.py NRF51 /dev/ttyACM0
```

carlescufi, 2017-04-12:
Also this issue should be in pc-ble-driver-py and  not in pc-ble-driver since you seem to be using the Python wrappers.

budius, 2017-04-13:
hi @carlescufi 

Thanks a lot for your help.
Yesterday I went knee deep on this and I got it working. As far as I can see it's basically the same as you did. From a macro level it was.
- started with a fresh PI install (actually a PI computer module)
- sudo apt-get install a bunch of stuff
- clone recursive pc-ble-driver-py.
- download and build boost
- cmake/make pc-ble-driver
- cmake/make python wrapper
- `python setup.py install`
- execute `heart_rate_collector.py` and we could see it scanning the bluetooth

It seems my main mistake when opening this ticket was to use `pip install`.
I'll save your cmd-line history here as reference as it will be very helpful.

Since this is fixed we have new issues, but I'll redirect them to the devzone.
Thanks and closing the issue.

closed 2017-04-13 by budius