NordicSemiconductor/pc-ble-driver #39: Build fails on OSX

mifi, 2017-04-28:
Followed instructions:
https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/Installation.md#building-boost

Tried building boost with
```
./bootstrap.sh
./b2 address-model=64 link=static --with-thread --with-system --with-regex --with-date_time --with-chrono
```

Also tried with boost isstalled from brew:
`brew install homebrew/versions/boost155 --universal`
Also tried newest version:
`brew install boost`

All boost options give the same error when building pc-ble-driver.

OSX 10.12.3

```
$ xcodebuild -version
Xcode 8.2.1
Build version 8C1002

$ cmake --version
cmake version 3.6.3

CMake suite maintained and supported by Kitware (kitware.com/cmake).

$ export BOOST_ROOT=/Users/mifi/Desktop/nordic/boost_1_63_0
$ mkdir build
$ cd build
$ cmake -G "Unix Makefiles" ..
$ make
```

Error:
```
Scanning dependencies of target test_uart
[ 98%] Building CXX object CMakeFiles/test_uart.dir/test/test_uart.cpp.o
[ 99%] Linking CXX executable test_uart
ld: warning: ld: warning: ld: warning: ignoring file /Users/mifi/Desktop/nordic/boost_1_63_0/stage/lib/libboost_thread.a, file was built for archive which is not the architecture being linked (i386): /Users/mifi/Desktop/nordic/boost_1_63_0/stage/lib/libboost_thread.aignoring file /Users/mifi/Desktop/nordic/boost_1_63_0/stage/lib/libboost_regex.a, file was built for archive which is not the architecture being linked (i386): /Users/mifi/Desktop/nordic/boost_1_63_0/stage/lib/libboost_regex.aignoring file /Users/mifi/Desktop/nordic/boost_1_63_0/stage/lib/libboost_system.a, file was built for archive which is not the architecture being linked (i386): /Users/mifi/Desktop/nordic/boost_1_63_0/stage/lib/libboost_system.a

ld: warning: ignoring file /Users/mifi/Desktop/nordic/boost_1_63_0/stage/lib/libboost_chrono.a, file was built for archive which is not the architecture being linked (i386): /Users/mifi/Desktop/nordic/boost_1_63_0/stage/lib/libboost_chrono.a
ld: warning: ignoring file /Users/mifi/Desktop/nordic/boost_1_63_0/stage/lib/libboost_date_time.a, file was built for archive which is not the architecture being linked (i386): /Users/mifi/Desktop/nordic/boost_1_63_0/stage/lib/libboost_date_time.a
Undefined symbols for architecture i386:
  "boost::detail::thread_data_base::~thread_data_base()", referenced from:
      boost::detail::thread_data<boost::_bi::bind_t<unsigned long, boost::_mfi::mf0<unsigned long, boost::asio::io_service>, boost::_bi::list1<boost::_bi::value<boost::asio::io_service*> > > >::~thread_data() in test_uart.cpp.o
  "boost::system::system_category()", referenced from:
      boost::asio::error::get_system_category() in test_uart.cpp.o
      boost::system::error_code::error_code() in test_uart.cpp.o
      boost::thread_exception::thread_exception(int, char const*) in test_uart.cpp.o
      ___cxx_global_var_init.2 in test_uart.cpp.o
  "boost::system::generic_category()", referenced from:
      ___cxx_global_var_init in test_uart.cpp.o
      ___cxx_global_var_init.1 in test_uart.cpp.o
  "boost::thread::start_thread_noexcept()", referenced from:
      boost::thread::start_thread() in test_uart.cpp.o
  "boost::thread::detach()", referenced from:
      boost::thread::~thread() in test_uart.cpp.o
  "typeinfo for boost::detail::thread_data_base", referenced from:
      typeinfo for boost::detail::thread_data<boost::_bi::bind_t<unsigned long, boost::_mfi::mf0<unsigned long, boost::asio::io_service>, boost::_bi::list1<boost::_bi::value<boost::asio::io_service*> > > > in test_uart.cpp.o
  "vtable for boost::detail::thread_data_base", referenced from:
      boost::detail::thread_data_base::thread_data_base() in test_uart.cpp.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
ld: symbol(s) not found for architecture i386
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [test_uart] Error 1
make[1]: *** [CMakeFiles/test_uart.dir/all] Error 2
make: *** [all] Error 2
```

bihanssen, 2017-04-28:
Hi! This looks to be related to boost and the correct architecture not being available. Can you try doing `brew uninstall --force boost` before doing `brew install homebrew/versions/boost155 --universal`.
The newest version of brew will not work because of a change the made after version boost155.
The --force uninstall is important, I will add it to the README if you can confirm that it works for you.

mifi, 2017-04-28:
i tried
```
brew uninstall --force boost
brew install homebrew/versions/boost155 --universal
export BOOST_ROOT=/usr/local/Cellar/boost\@1.55/1.55.0_1/
...
make
```

Same error still

bihanssen, 2017-04-28:
Ok, I will try to reproduce the issue and report back.

bihanssen, 2017-04-28:
I was able to reproduce the problem. It seems brew lately has removed completely the version of boost that is able to build with --universal (32&64 bit). This means that we might have to fall back to compiling boost from source, or using one single architecture instead of combined 32&64. I will look more into the subject.

bihanssen, 2017-05-03:
Hi, 
I removed the building of 32 bit binaries and it fixes the problem with boost. Could you confirm that you are able to build after pulling the latest changes? Also, now you could use the default brew boost version.

mifi, 2017-05-03:
I can confirm that it works now. shall i make an issue in pc-ble-driver-js to update the commit reference and create new version?

bihanssen, 2017-05-03:
I have updated the commit reference in pc-ble-driver-js, and also tested that it works with nRF Connect. We will tag it with a new version number at some point later, for now it's just in master.
Thanks for your feedback.

closed 2017-05-03 by bihanssen