NordicSemiconductor/pc-ble-driver #42: Any plan to release the pc-ble-driver for SDK13?

Arepas, 2017-05-23:
Hi guys any plan for release a compatible pc-ble-drver for SDK13?

I tried doing some patch to the connectivity firmware, but it requires some change at pc-ble-driver level too.

bencefr, 2017-05-24:
Hi @Arepas, I would think we will provide such driver however anyone who could give you the correct answer is currently unavailable. I'll follow up.

bencefr, 2017-05-29:
@Arepas , the answer is yes, we definitely do plan to support latest SDK, however currently we are a bit under-powered and also have other priorities to focus on. Development will pick up much more momentum after the summer vacations are over.

bartholmberg, 2017-08-23:
Any updates on the new driver for SDK 13?

bihanssen, 2017-08-24:
@bartholmberg @Arepas Work for adding support for latest relevant SDK has been planned in and will start shortly. Expect it to take around a month give or take, depending on other activities.

closed 2017-08-24 by bihanssen