NordicSemiconductor/pc-ble-driver #50: Queued Writes (missing `sd_ble_user_mem_reply` implementation)

torfinnberset, 2017-09-23:
I am trying to implement queued writes with the connectivity firmware. However, the required `sd_ble_user_mem_reply` is not fully implemented, and returns `NRF_ERROR_INVALID_PARAM` if I try to pass a memory block allocated on the host application side. 

https://github.com/NordicSemiconductor/pc-ble-driver/blob/4bda53f1ad73b0ba55e02842052e1843c69aea1f/src/common/ble_impl.cpp#L223

This return value is not specified in `ble_impl.h`, but I assume this is because the SoftDevice would fill in the memory block on the nRF target, but nevertheless I would like to have a workaround for this.

ThomasGerstenberg, 2017-09-25:
@torfinnberset This event is a request for a block of memory within the nRF Target's application ram space that the Softdevice can use as a temporary buffer/scratchpad to store the queued writes before execution. Sending it a pointer to a block of memory on the host doesn't really make sense, since the softdevice has no knowledge of this memory space (nor can write to it directly), and the Connectivity firmware itself does not handle any of this functionality, as it shouldn't.
What you'll need to do is reply with `sd_ble_user_mem_reply(NULL)`, and handle all of the queued writes in host application itself.

What this will entail is allocating some buffer application-side in order to build out a list of (handle, offset, data) for each `BLE_GATTS_OP_PREP_WRITE_REQ`. The target does not need to know about it, so reply NULL for the user memory request (you'll handle it yourself). Then, on `BLE_GATTS_OP_EXEC_WRITE_REQ_NOW` iterate through the list and update the characteristics that were written to.

See also: 
[Message sequence for handling queued writes yourself](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s132.api.v3.0.0%2Fgroup___b_l_e___g_a_t_t_s___q_u_e_u_e_d___w_r_i_t_e___n_o_b_u_f___n_o_a_u_t_h___m_s_c.html)
I recently got queued writes working, see [these](https://github.com/ThomasGerstenberg/blatann/blob/master/blatann/gatts.py#L227-L229). [lines](https://github.com/ThomasGerstenberg/blatann/blob/master/blatann/gatts.py#L172-L179) which handle the queued writes and execution (Python bindings)

torfinnberset, 2017-09-25:
@ThomasGerstenberg thanks for the detailed explanation. Given the message sequence chart, I think I can indeed implement this myself, application side. 

For other users of the `pc-ble-driver`, it might be worthwhile to update the documentation for `sd_ble_user_mem_reply` to avoid confusion.

bihanssen, 2017-09-29:
Thanks @ThomasGerstenberg, your explanation is spot on. Functions that handle memory locations in the SoftDevice API behave differently when used over Serialization, as memory is split between two physically different devices.

It is also true that is fully possible to accomplish queued writes over serialization, by calling sd_ble_user_mem_reply(NULL).

As a side note I can also add that queue write is implemented in pc-ble-driver-js, so it could be worthwhile to look at that for reference: [mem_reply](https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/master/api/adapter.js#L1925), [gattcwrite](https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/master/api/adapter.js#L1508), [gattswrite](https://github.com/NordicSemiconductor/pc-ble-driver-js/blob/master/api/adapter.js#L1761).

closed 2017-09-29 by bihanssen