NordicSemiconductor/pc-ble-driver #54: Hi guys i met errors while set up for my device.

KwonGiho, 2017-10-26:
Hi genius guys.
I’m a junior developer who is making DFU software for windows 10. I encountered new errors while setting up the following Installation.md

Attached below are the photos and explanation on how I approached to build this repository. 
The first error was encountered while using cmake. 
[image]
I followed the steps of [windows-2] (https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/Installation.md#windows-2)

My device environment is Windows 10 x64bit, using Visual Studio 15 2017. It was a success when I build it without “Win64” keyword but there was a new warning/error message about “Architecture not set, using native…”  
I would like to know what does that warning message mean and what could be done to solve it? 
Anyway, I have proceeded with the next step while ignoring the previous warning message by typing msbuild ALL_BUILD.vcxproj /p:Configuration=Debug
[image]
And Error 3 appeared. I was thinking if this Error 3 was the result of Error 1? Could you please recommend or suggest what can I do to build this repository? 
Thank you.

bihanssen, 2017-10-26:
Hi @KwonGiho 
It is important to have Boost compiled correctly in the same architecture as you compile pc-ble-driver in. Your first error seems to indicate you have run cmake before with a different setting. You should delete all files in the build folder and run cmake again.
The second error window might indicate mismatch in architecture of compiled Boost and pc-ble-driver.

You can't mix 64 bit Boost and 32 bit pc-ble-driver, so make sure they are the same. I suggest rebuilding both in 32 bit.

KwonGiho, 2017-10-27:
I’m already set up all of them.

But still not operating yet too T^T

By the way , I solve it on MacOS but there is has an issue on MacOS.

I think this library require J-Link Device to connect each other right? ( I think that Windows need J-Link too , right? )

I can’t connect both of device without J-Link device?

Thanks for reading.

bihanssen, 2017-12-20:
Hi,
pc-ble-driver requires an nRF development kit or dongle with JLink debugger. 

When you say *both* devices, what devices do you mean? The peer device (the remote device connected to over BLE) can be any BLE device, no requirement for JLink there.

bencefr, 2018-02-09:
Should this issue be still relevant, feel free to reopen it.
Ask questions on [DevZone](https://devzone.nordicsemi.com/)

closed 2018-02-09 by bencefr