NordicSemiconductor/pc-ble-driver #56: Can't connect to 1m baud version of the 1.2.1 connectivity firmware.

rmcgregor1990, 2017-10-27:
I cannot successfully open an RPC connection to the 1M baud version of the 1.2.1 connectivity firmware.
The 115k2 version seems to work fine.

I'm using your heart_rate_monitor example built with 1M and 115k2 bauds respectively.

I was previously using version 1.1.0 and I had no problems.

Thanks,
Rob

bihanssen, 2017-10-27:
Hi,
what kit are you using, and what sd_api version?

rmcgregor1990, 2017-10-27:
I'm using the connectivity firmware builds included in the v2.1.1 release for API v3.

bihanssen, 2017-10-27:
Could you also specify the kit (nRF52 devkit, nRF51 dongle?), and operating system. We need the info for reproducing the issue. Thanks.

rmcgregor1990, 2017-10-28:
I'm using a nRF52 Dev kit on windows 7. I'm building 64 bit bit static lib using both the visual studio 14 compiler and mingw. It's build against boost 1.63.  I'm using cython to create a 2.7/3.6 compatible python package for testing BLE interfaces. 

I've been building against an older version of the repo up to now and decided to update to the v2.1.1 release in the hope of fixing some issues I'm having for MTU sizes greater than 60.

I couldn't connect to the 1m baud version of the 1.2.1 firmware however. I've verified this with your heart rate monitor example build for 1m baud under VS14 for Windows 64 bit.

bihanssen, 2017-10-28:
Thanks for the details and for reporting, we will look into this during the coming week.

rmcgregor1990, 2017-10-30:
This might be helpful. 

I diffed the two firmware .hex builds, they differ by 3 lines. 
Found the line that corresponded to the UART baud-rate change (0x01D7E000 for 115k2 and 0x10000000 for 1m),
I then changed that single line in the 115k2 image to use 1m. 
This modified image now seems to work fine when connected too with 1m baud-rate.

bihanssen, 2017-11-03:
Thanks for the info. The connectivity hex v1.2.1 1m had increased retransmission timeout values in the transport layer, whereas the 115k2 had the same timeout value as before. It seems this increase in value has caused issues, which is odd since it passed the release tests on the same setup. We didn't manage to look into the issue this week, but it's on our list.

bihanssen, 2017-12-20:
Hi, @rmcgregor1990, 
a new version of connectivity firmware was recently added, which addresses issues related to transport connection establishment. Hopefully this would solve the issues you were also seeing.
The hex files are available in tag v2.2.0.

bencefr, 2018-02-09:
Should this issue still be relevant, feel free to reopen it.

closed 2018-02-09 by bencefr