NordicSemiconductor/pc-ble-driver #67: I am facing error building from the source

mohamed-elsabagh, 2018-02-06:
Hello guys, I am trying to build the pc-ble-driver library from source on an Ubuntu machine and I am following the steps in the README file
I installed boost and cloned the library run cmake, and when I run make it gives me the following errors
```
/home/myMachine/Desktop/test/pc-ble-driver-py/pc-ble-driver/include/common/internal/transport/uart_boost.h:100:18: error: ‘io_service’ in namespace ‘boost::asio’ does not name a type
     boost::asio::io_service ioService;
```

I am trying to run this also on a RPI3 and I got the same error

bencefr, 2018-02-07:
Hi Mohamed, did you download Boost or installed it from ubuntu repositories? What is your ubuntu version, and can you please check boost version as well?

mohamed-elsabagh, 2018-02-07:
I followed those steps mentioned here https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/Installation.md#building-boost
I downloaded a tar file from this link http://www.boost.org/users/download/ and I am running Ubuntu 17.10

bencefr, 2018-02-07:
It looks like we have an issue with the recent version of boost, thank you for reporting it.
While we work out a fix, I suggest please install [boost 1.65.1](https://dl.bintray.com/boostorg/release/1.65.1/source/) which - I have just checked - still works fine.

mohamed-elsabagh, 2018-02-09:
Thanks so much for your help, now I am using Boost 1.63.0 and cmake 3.9.1 and I am getting the following error ```Could NOT find Threads (missing: Threads_FOUND)```

closed 2018-02-09 by bencefr