NordicSemiconductor/pc-ble-driver #69: Serial Transport event handler crashes after few hours when running the example hrc_v2 on Visual Studio 14.0 win64

angsTD, 2018-02-12:
I have been running the hrc_v2, with the application scannning and printing advertisement reports, the application crashes due to "deque iterator not dereferencable" issue

 	msvcp140d.dll!000007fed7200806()	Unknown
 	hrc_v2.exe!std::_Deque_const_iterator<std::_Deque_val<std::_Deque_simple_types<eventData_t> > >::operator*() Line 330	C++
 	hrc_v2.exe!std::_Deque_iterator<std::_Deque_val<std::_Deque_simple_types<eventData_t> > >::operator*() Line 591	C++
 	hrc_v2.exe!std::deque<eventData_t,std::allocator<eventData_t> >::front() Line 1438	C++
 	hrc_v2.exe!std::queue<eventData_t,std::deque<eventData_t,std::allocator<eventData_t> > >::front() Line 133	C++
>	hrc_v2.exe!SerializationTransport::eventHandlingRunner() Line 167	C++
 	[External Code]

bencefr, 2018-02-13:
@angsTD , thank you for reporting, we are investigating the issue.

bencefr, 2018-02-14:
@angsTD , I believe we have fixed by #70.

closed 2018-02-14 by bencefr