NordicSemiconductor/pc-ble-driver #74: HRM example does not work with MTU != 23

torfinnberset, 2018-04-01:
I am trying to get longer MTU sizes working with pc-ble-driver and SoftDevice v5. I am using the hrm_v5 example as a starting point. 

If I add `#define GATT_MTU_SIZE_DEFAULT 23` at the top of `main.c`, I get the following output:

```
/Users/torfinn/nrf-pc-ble-driver/examples/heart_rate_monitor/cmake-build-debug/hrm_v5 /dev/cu.usbmodem1411
Serial port used: /dev/cu.usbmodem1411
Baud rate used: 115200
Info: Successfully opened /dev/cu.usbmodem1411. Baud rate: 115200. Flow control: none. Parity: none.

Status: 6, message: Target Reset performed
Status: 7, message: Connection active
Advertising data set
Services initiated
Characteristics initiated
Started advertising
Connected, connection handle 0x0000
MTU exchange request reply failed. Error code: 0x07
Disconnected
```

If I leave it at the default 23, everything works just fine. I am using the nRF52832 development kit, flashed with SoftDevice v5 connectivity firmware from the `/hex` folder.

On a different note, it would be very useful for end-users if the example from https://devzone.nordicsemi.com/b/blog/posts/throughput-and-long-range-demo could be ported to pc-ble-driver. Even better would be SoftDevice v6 support and examples!

bihanssen, 2018-04-17:
You write that you add #define GATT_MTU_SIZE_DEFAULT 23 to main, but do you mean a value _other_ than 23, like stated in the headline?

bihanssen, 2018-04-17:
Ok. Are you using pc-ble-driver HRM & HRC together, or some other central?

bihanssen, 2018-04-17:
I believe the problem here is that it is not allowed to set the ATT MTU higher than what the peer supports. You need to set the value same or lower than what is indicated in the received event [client_rx_mtu](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/src/sd_api_v5/sdk/components/softdevice/s132/headers/ble_gatts.h#L390). The example should have shown this to make it more clear.

bihanssen, 2018-05-18:
@torfinnberset I found an issue in the HRM example that might be related to the issue you were seeing. Does the fix in PR #102 solve the MTU issue for you?

torfinnberset, 2018-05-18:
@bihanssen I can check next week! Thanks for looking into it

torfinnberset, 2018-06-04:
Hi @bihanssen, I didn't get to it before now, but I am having issues with the branch in PR #102: 

```
TorfinnMBP:build torfinn$ ./hrm_v6 /dev/cu.usbmodem1441 
Serial port used: /dev/cu.usbmodem1441
Baud rate used: 115200
Info: Successfully opened /dev/cu.usbmodem1441. Baud rate: 115200. Flow control: none. Parity: none.

Status: 6, message: Target Reset performed
Failed to open nRF BLE Driver. Error code: 0x0D
```

I programmed the nRF52 Development Kit with the following commands with the hex files from the same branch (in /hex folder):

```
TorfinnMBP:sd_api_v6 torfinn$ nrfjprog -f nrf52 -e
Erasing user available code and UICR flash areas.
Applying system reset.

TorfinnMBP:sd_api_v6 torfinn$ nrfjprog -f nrf52 --program connectivity_6.0_115k2_with_s132_6.0.hex 
Parsing hex file.
Reading flash area to program to guarantee it is erased.
Checking that the area to write is not protected.
Programing device.
```

I will also try the hex files from the other branch (`add/hex-filex`) to see if that fixes it

torfinnberset, 2018-06-04:
@bihanssen Ok, it seems the hex files from `add/hex-files` works for me. Also the example works for MTUs other than 23, so the original issue can be closed

jackadrianglass, 2018-07-12:
@torfinnberset where is the other branch ( add/hex-files )? I'm getting the same problem where the device fails to open with error code 0x0D.

torfinnberset, 2018-07-20:
@jackadrianglass probably this branch is gone by now, you can try the latest from `master`

closed 2018-07-20 by torfinnberset