NordicSemiconductor/pc-ble-driver #77: not able to run-pc-ble-driver with nodejs binding

tonykappen, 2018-04-05:
I downloaded the pc-ble-driver from https://github.com/NordicSemiconductor/pc-ble-driver. Then I flashed the connectivity_2.0.1_115k2_with_s130_2.0.1.hex using the command “nrfjprog -f NRF52 –program connectivity_2.0.1_115k2_with_s130_2.0.1.hex”. I have a nRF52(PCA 10040) development kit. 
[image]
I got the above result after flashing the connectivity hex file. Now I am under the assumption that the connectivity hex file is correctly flashed into the development kit. I am also assuming in this situation that the connectivity hex file plays the role of a softdevice in the development kit. Next, I want to flash the actual application firmware. I wish to run the Bluetooth heart rate example present in the nodric SDK folder. This is the path which I followed to flash the find the hex file “nRF5_SDK_14.2.0_17b948a\examples\ble_peripheral\ble_app_hrs\pca10040\s132\ses\Output\Debug\Exe”
[image]
An error is getting generated when I try to flash the application firmware hex file. My goal is to run the heart rate monitor example present in the pc-ble-driver-js using the node command. The example can be found in the URL https://github.com/NordicSemiconductor/pc-ble-driver-js/tree/master/examples. The command which I am going to use to run the example is “node heart_rate_monitor.js”. Our end goal is to develop a nodejs application making calls to nRF52 chip. 
I also tried flashing the pc-ble-driver with nRFgo studio. I am trying to flash the connectivity firmware using the program softdevice tab.
[image]
I am getting the error as shown in the above screen shot. Next, I try to flash the connectivity firmware using Program Application tab. Now the nRF52 dev kit is getting programmed.
[image]
Please refer the last line of the log file. I assume that that the connectivity hex file got flashed as an application firmware. After this, I flash the application firmware in the following path “nRF5_SDK_14.2.0_17b948a\examples\ble_peripheral\ble_app_hrs\pca10040\s132\ses\Output\Debug\Exe”, the development kit gets flashed. Please refer the below screenshot. 
[image]

The PC gets connected for some time and then it goes back the paired state. After that the PC shows that the development kit goes back to the paired state. When I run the node command, I get the following error. Please refer the below screenshot. 

[image]

bencefr, 2018-04-06:
Hi @tonykappen ,

The connectivity hex file is an application firmware that also contains the softdevice.
The idea is that a DK running the connectivity firmware talks on UART to the host computer and exposes the softdevice's API by which you can turn your nRF chip into a any kind of BLE device.

E.g. the heart-rate-collector and heart-rate-monitor examples under pc-ble-driver(-js) are only using the connectivity hex file, there's no extra HRC/HRM application hex files. If you have two DevKits, turn one into HRC, the other HRM.

The examples under SDK are applications running on nRF chips, they don't use the connectivity.

So after you flash the correct connectivity you should be able to run those examples. Make sure for nRF52832/PCA10040 you use the hex file from sd_api_v5 (with s132) and for the baudrate you will expect from your application (you can change the source code as well).

bencefr, 2018-05-18:
Closing due to inactivity, please reopen if this is still an issue.

closed 2018-05-18 by bencefr