NordicSemiconductor/pc-ble-driver #78: issue in compiling the connectivity .hex files

tonykappen, 2018-04-06:
I am trying to run the pc-ble-driver on a nRF52 (PCA10040) development kit. I downloaded the pc-ble-driver from https://github.com/NordicSemiconductor/pc-ble-driver. Then I am following the installation steps in the https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/Installation.md. Then I proceeded to the “Compiling the connectivity .hex files” in the webpage. I ran the bootstrap file by double clicking on the bootstrap section. Please refer the below screenshot.  
[image]

In this step I assume that softdevice got downloaded and the patch was applied. I also assume that the the step1 and step 2 was finished downloading. I could not find the “ble_connectivity” folder under the “ble_central_and_peripheral”. Please refer the below screenshot. 

[image]

I saw the “ble-connectivity” folder under the “connectivity” folder. Please refer the below screen shot. 

[image]
When I try to merge the hex file, I am getting the following error 

[image]
How can I get over this issue?

tonykappen, 2018-04-06:
Nordic also released a new version of SDK with the version 15. Any advice how to go go about it ? Also I use Segger IDE.

bencefr, 2018-04-06:
Do you use a custom DK? Normally you don't need to compile the connectivity file, the nRF52 PCA10040 should be working with the pre-compiled one.

In the screenshot I see you try to merge a connectivity hex file (which does contain s132 softdevice) with a softdevice, they are for this reason conflicting.

For SDK 15, the softdevice API v6 is not fully ready for pc-ble-driver, there is a work in progress branch.

bencefr, 2018-05-18:
Closing due to inactivity, please reopen if this is still an issue.

closed 2018-05-18 by bencefr