NordicSemiconductor/pc-ble-driver #93: Fail to run HRC example of branch v6-upgrade on mt7628 chip(linux platform)

lyx10101010, 2018-04-25:
I have already successfully run HRC example of V2 and V5 on mt7628( MIPS linux platform), but the following errors occurred when running V6‘s HRC example:
====================
Serial port used: /dev/ttyUSB0
Baud rate used: 1000000
Info: Successfully opened /dev/ttyUSB0. Baud rate: 1000000. Flow control: none. Parity: none.

Status: 6, message: Target Reset performed
Status: 7, message: Connection active
Status: 4, message: Error sending packet to target. Code #13
Scan start failed with error code: 3
====================
Q: How to fix this issue?

more detail information:
linux kernel version : 3.18.29  
nordic chip : nrf52832
connectivity_6.0_1m_with_s132_6.0.hex was run on nrf52832

bihanssen, 2018-04-25:
Hi, if you have not disabled MSD on the jlink device, you might run into a known [jlink/osx/linux issue](http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.tools%2Fdita%2Ftools%2FnRF_Connect%2FnRF_Connect_Troubleshooting.html). (It's called OSX issue because it happened there first, and more recently on  linux).  It would appear only in v6 scan start because of an API change in SD causing longer packets.

bihanssen, 2018-04-25:
I was assuming you were using a devkit from Nordic which uses jlink chip for usb-cdc uart, like pca10040. Is this the board you are using?

Provided that MSD is the issue here:
1) The error caused by the jlink device is that when MSD is enabled, it will truncate usb packets above a certain threshold, causing the long packets to be invalid. This does not happen when MSD is disabled.
2) To disable MSD from Linux, run jlinkexe binary (from segger), and in that console, type `msddisable` and power cycle the device.

lyx10101010, 2018-04-25:
We are not test the example on devkit board. In addition, our segger Jlink device is on windows platform.

bihanssen, 2018-04-25:
Ok, can you describe the board you are using, if you are not using the Nordic devkit? What chip is hosting the USB-CDC, if not JLink on the Nordic devkit?
Regarding jlink and windows, you would need to use the JLinkCommander tool to get the jlink terminal.

lyx10101010, 2018-04-26:
1. nrf52832(Minimum system board)  UART<===>USB2Serial(cp2104 chip)<===>mt7628's USB.
2.I have already successful disable msd,but the error still occurred.

bihanssen, 2018-04-30:
I see, you have a custom setup without JLink involved. You could increase the log level to SD_RPC_LOG_DEBUG in HRC  example to get more info on where it is failing. 
Maybe the USB2Serial chip has the same issue as the JLink device, in which case a USB trace would be helpful for debugging, with USBLyzer or similar tool.

lyx10101010, 2018-05-03:
After set the log level to SD_RPC_LOG_DEBUG.
The log messages is as follows:
====================log start===============
Serial port used: /dev/ttyUSB0
Baud rate used: 1000000
Info: Successfully opened /dev/ttyUSB0. Baud rate: 1000000. Flow control: none. Parity: none.

Log: State change: STATE_START -> STATE_RESET

Log:        1 ->  []
               type:          RESERVED_5 reliable: no seq#:0 ack#:0 payload_length:0 data_integrity:0 err_code:0
Status: 6, message: Target Reset performed
Log:     1/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
Log: State change: STATE_RESET -> STATE_UNINITIALIZED

Log:        2 ->  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
Log:     2/ 0 <-  [02 7d ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC_RESP]
Log: State change: STATE_UNINITIALIZED -> STATE_INITIALIZED

Log:        3 ->  [03 fc 11 ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0
               [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
Log:     3/ 0 <-  [01 7e ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC]
Log:        4 ->  [02 7d ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0
               [SYNC_RESP]
Log:        5 ->  [03 fc 11 ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0
               [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
Log:     4/ 0 <-  [04 7b 11 ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0
               [CONFIG_RESP [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
Log: State change: STATE_INITIALIZED -> STATE_ACTIVE

Status: 7, message: Connection active
Log:        6 ->  [00 69 40 00 00 00 01 00 01 00 01 00 ]
               type:     VENDOR_SPECIFIC reliable:yes seq#:0 ack#:0 payload_length:c data_integrity:1 header_checksum:72 err_code:0
Log:        7 ->  [00 69 40 00 00 00 01 00 01 00 01 00 ]
               type:     VENDOR_SPECIFIC reliable:yes seq#:0 ack#:0 payload_length:c data_integrity:1 header_checksum:72 err_code:0
Log:        8 ->  [00 69 40 00 00 00 01 00 01 00 01 00 ]
               type:     VENDOR_SPECIFIC reliable:yes seq#:0 ack#:0 payload_length:c data_integrity:1 header_checksum:72 err_code:0
Log:     5/ 0 <-  [03 fc 11 ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0
               [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
Log:        9 ->  [04 7b 11 ]
               type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0
               [CONFIG_RESP [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
Log:       10 ->  [00 69 40 00 00 00 01 00 01 00 01 00 ]
               type:     VENDOR_SPECIFIC reliable:yes seq#:0 ack#:0 payload_length:c data_integrity:1 header_checksum:72 err_code:0
Log:     6/ 0 <-  []
               type:                 ACK reliable: no seq#:0 ack#:1 payload_length:0 data_integrity:0 err_code:0
Log:     7/ 0 <-  [01 69 00 00 00 00 ]
               type:     VENDOR_SPECIFIC reliable:yes seq#:0 ack#:1 payload_length:6 data_integrity:1 header_checksum:ca err_code:0
Log:       11 ->  []
               type:                 ACK reliable: no seq#:0 ack#:1 payload_length:0 data_integrity:0 err_code:0
Log:       12 ->  [00 69 23 00 00 00 01 01 96 00 ]
               type:     VENDOR_SPECIFIC reliable:yes seq#:1 ack#:1 payload_length:a data_integrity:1 header_checksum:89 err_code:0
Log:     8/ 0 <-  []
               type:                 ACK reliable: no seq#:0 ack#:2 payload_length:0 data_integrity:0 err_code:0
Log:     9/ 0 <-  [01 69 00 00 00 00 ]
               type:     VENDOR_SPECIFIC reliable:yes seq#:1 ack#:2 payload_length:6 data_integrity:1 header_checksum:c1 err_code:0
Log:       13 ->  []
               type:                 ACK reliable: no seq#:0 ack#:2 payload_length:0 data_integrity:0 err_code:0
Log:       14 ->  [00 60 ]
               type:     VENDOR_SPECIFIC reliable:yes seq#:2 ack#:2 payload_length:2 data_integrity:1 header_checksum:0 err_code:0
Log:    10/ 0 <-  []
               type:                 ACK reliable: no seq#:0 ack#:3 payload_length:0 data_integrity:0 err_code:0
Log:    11/ 0 <-  [01 60 00 00 00 00 ]
               type:     VENDOR_SPECIFIC reliable:yes seq#:2 ack#:3 payload_length:6 data_integrity:1 header_checksum:b8 err_code:0
Log:       15 ->  []
               type:                 ACK reliable: no seq#:0 ack#:3 payload_length:0 data_integrity:0 err_code:0
Log:       16 ->  [00 8a 01 00 01 01 00 00 00 00 00 a0 00 50 00 00 00 01 c8 26 41 00 64 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0 [... cut ...]
Log:       21 ->  [00 8a 01 00 01 01 00 00 00 00 00 a0 00 50 00 00 00 01 c8 26 41 00 64 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ]
               type:     VENDOR_SPECIFIC reliable:yes seq#:3 ack#:3 payload_length:7d data_integrity:1 header_checksum:40 err_code:0
Status: 4, message: Error sending packet to target. Code #13
Scan start failed with error code: 3
==================log end==========================

bihanssen, 2018-05-03:
The log shows that everything looks normal until the scan_start command is issued (`00 8a ..`). scan_start is never ACKed by the connectivity firmware. It may still point to that the whole packet has not been received by connectivity due to corrupted packet, possibly due to USB issue as described. 
Things to try out: 
- Send a shorter buffer in scan_start, as a workaround (reduce [mp_data](https://github.com/NordicSemiconductor/pc-ble-driver/blob/feature/v6-upgrade/examples/heart_rate_collector/main.c#L125) length).
- Try to compile connectivity and set breakpoint at packet reception to see if the whole packet is received.

bihanssen, 2018-05-18:
Closing the issue due to inactivity. Consider posting on devzone for further assistance.

closed 2018-05-18 by bihanssen