NordicSemiconductor/pc-ble-driver #110: Error encountered during discovery of primary services

timeye, 2018-06-11:
Error encountered during discovery of primary services
error_code = 0x03    returned from sd_ble_gattc_primary_services_discover() 
It appears that this error_code points to some kind of  NRF_ERROR_INTERNAL

Debug log is attached...

Does anybody have a handy dandy look-up chart for decoding these RPC Log Entries??? I can decode a few using the file h5_transport.cpp...  but mixed in with these are BLE event codes and a few that I haven't identified...

Any suggestions would be greatly appreciated!  thx...

[VS_runtime_logfile.txt](https://github.com/NordicSemiconductor/pc-ble-driver/files/2089302/VS_runtime_logfile.txt)

bencefr, 2018-06-20:
Hi @timeye , can I ask you to describe the environment in a bit more detail? Which version/branch have you used and with what devices?

timeye, 2018-06-21:
I am using the latest pc-ble-driver with feature/v6-upgrade.  Visual Studio 2013 Pro. Boost 1.67.0
For the Connectivity portion, I am using an 52832 Development kit and the Nordic SDK v15.0.0.
For the BLE Peripheral, I am using a 52840 Development kit and the Nordik SDK v15.0.0.

I have run the example code (heart_rate_collector with the pc-ble-driver) and the HRM BLE Peripheral example firmware on the BLE Peripheral 52840.  The example seems to run fine.

When I use my custom driver and custom BLE peripheral code, what I see is shown in the logfile above.
I have decoded the RPC Log events a bit...  If you look at the logfile:
The Connect is shown on Line 146. 
Then I receive a Connected event on Line 158.
I proceed to the Discovering Services on Line 167......
but then I continue to receive Connected events on Lines 170, 176, 182, & 188

This almost seems like there's a Response ---> Ack timer that needs to be adjusted, since I am getting multiple repeated Connected events

kenr, 2018-09-20:
Hi @timeye, can you provide the source code for the pc-ble-driver based application?

timeye, 2018-09-23:
Hello @kenr, this is a proprietary app, so an NDA would be required...

kenr, 2018-10-08:
Hi @timeye, you can contact your regional sales manager to get an NDA. We can pick up it up from there afterwards. Please go through DevZone and Nordic Support portal for further investigation of this issue.