NordicSemiconductor/pc-ble-driver #112: Writing more than 20 bytes to peripheral crashes driver

torfinnberset, 2018-06-19:
I am using SoftDevice v5 on commit 971241c19abc33bbc6469e1be601a2ab78fbb35b. 

Whenever I, as Central, try to write more than 20 bytes to the connected peripheral, I get the following errors from driver:

```
Warning: Failed to receive response for command
Status: 4, message: Error sending packet to target. Code #3
status_handler: PKT_SEND_ERROR
NRF_ERROR_INTERNAL
```

I have already negotiated an MTU of 185, and I am using Data Length extension. Receiving more than 20 bytes from the peripheral is no problem.

These are the flags I am using:

```
const ble_gattc_write_params_t write_params = {
    .write_op = BLE_GATT_OP_WRITE_CMD,
    .flags    = BLE_GATT_EXEC_WRITE_FLAG_PREPARED_WRITE,
    .handle   = handle,
    .offset   = 0,
    .len      = 21,
    .p_value  = p_buffer
};

sd_ble_gattc_write(conn_handle, &write_params);

<error here>
```

bencefr, 2018-06-20:
Thanks for reporting, internal issue has been created our backlog.

kenr, 2018-07-20:
@torfinnberset This is fixed in master now.

closed 2018-07-20 by kenr

torfinnberset, 2018-07-20:
@kenr Thanks for fixing. By the way I really appreciate the new way of working in this repo. Keep up the good work!