NordicSemiconductor/pc-ble-driver #113: Race condition in h5_transport.cpp

kleunen, 2018-06-20:
In the function 
uint32_t H5Transport::open(status_cb_t status_callback, data_cb_t data_callback, log_cb_t log_callback)

a race condition exists on starting the state machine. In this function the stateMachine is started:
startStateMachine();

Later, 

when all the open functions are completed, a notify_all is used to signal a background thread:

_exitCriterias->isOpened = true;
syncWaitCondition.notify_all();

However, no check is performed if the background thread is already waiting on the signal:
    while (!exit->isFullfilled())
        {
            syncWaitCondition.wait(syncGuard);
        }

If the notify_all occurs before 'syncWaitCondition.wait(syncGuard)' is called, the notify is missed. The signalling thread should wait until the background thread has called syncWaitCondition, before calling the notify_all.

kleunen, 2018-06-20:
I think the problem is isOpened is written in the exitCriterias without taking a mutex!
[L161](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/src/common/transport/h5_transport.cpp#L161)

This should be:
```c++
{
    std::unique_lock<std::mutex> syncGuard(syncMutex);
    _exitCriterias->isOpened = true;
}

syncWaitCondition.notify_all();
```

It seems this pattern is repeated a lot in the code and the exitCriterias are written many times without locking the mutex! 

For example:
[L145](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/src/common/transport/h5_transport.cpp#L145)
[L156](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/src/common/transport/h5_transport.cpp#L156)
[L318](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/src/common/transport/h5_transport.cpp#L318)
[L343](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/src/common/transport/h5_transport.cpp#L343)
[L387](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/src/common/transport/h5_transport.cpp#L387)
[L401](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/src/common/transport/h5_transport.cpp#L401)

kleunen, 2018-06-21:
I tried with the mutex, but this does not seem to resolve the issue. The only way I seem to resolve it now, is by adding a delay after 'startStateMachine()' [here](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/src/common/transport/h5_transport.cpp#L137)

```C++
startStateMachine();
std::this_thread::sleep_for(std::chrono::milliseconds(1000));
```

Not really a nice solution

kleunen, 2018-06-21:
I think I understand the problem now: is the exitStatus reset occurs after isOpened is set to true, the isOpened event is missed:

Reset in the background thread:
[Reset](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/src/common/transport/h5_transport.cpp#L483)

And isOpened in the foreground thread:
[isOpened](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/src/common/transport/h5_transport.cpp#L161)

if the order is L483 first and then L161, all is fine

if the order is L161 first and then L483, the [waitForState](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/src/common/transport/h5_transport.cpp#L164) will timeout, because the exitstatus is reset after setting isOpened to true. 

The whole 'exitCriterias' should be protected by a mutex, in the foreground thread and the background thread. And the whole switching between states should be protected by 'syncMutex' , because the previous state is finalized, and the new state is initialized, and inbetween your are not allowed to modify the state exitCriterias.

kenr, 2018-06-22:
Hi @kleunen,

Thank you for reporting your finding, this is something we need to look into.

You are right, value assignment to data members in classes inheriting ExitCriterias are not protected with a mutex, thereby we get race conditions as you describe.

Here is an illustration of one of the cases you described:

| time   |thread: "main" (L161)   | thread: stateMachineWorker (L487)   |
| ----- |:------------------------:| :-----------------------------------------|
| t0     |              |   while (!exit->isFullfilled())                 |
| t1     | _exitCriterias->isOpened = true | |
| t2     | syncWaitCondition.notify_all()    | |
| t3     |     |  syncWaitCondition.wait(syncGuard) **deadlock** |

I'll let you know when we have pushed a fix for this.

kleunen, 2018-06-22:
Hello Kenr, 

the scenario you described is not the scenario which causes the deadlock. The scenario is as follows:

| time | thread: "main" (L161) | thread: stateMachineWorker (L487) |
| ------------- |:-------------| :-----|
| t0   | _exitCriterias->isOpened = true | | 
| t1   | syncWaitCondition.notify_all() | | 
| t2   |  | exit->reset() | 
| t3   |  | while (!exit->isFullfilled()) | 
| t4   | if (waitForState(STATE_ACTIVE, OPEN_WAIT_TIMEOUT)) (results in timeout) |  |

kleunen, 2018-06-22:
It is better to require a proxy to force locking of some shared datatype. This way, the shared data type can never be accessed without requesting a mutex, forcing the programmer to use correct locking. For example an approach as this:

[Force locking for thread-safety](https://codereview.stackexchange.com/questions/15632/force-locking-for-thread-safety)

kenr, 2018-07-20:
@kleunen We improved the locking in critical sections in PR #116 . Let us know how it works for you.

closed 2018-07-20 by kenr

kleunen, 2018-07-23:
It works now, the serial port is now correctly opened.