NordicSemiconductor/pc-ble-driver #114: sd_ble_cfg_set returning INTERNAL_ERROR

ptagl, 2018-06-22:
Hello,

I am currently developing a Desktop Application based on the PC Ble Driver, release 3.0.1. In particular I'm using the shared library DLL v5, since I need to connect to a nRF52 Development Kit and use it like a Central device.

In order to make my C++ application work, I flashed the S132 hex file into the Development Kit (v5, 1M baudrate) and used the Collector example to make the initialization.

I get an error sometimes (randomly) when I try to configure some parameters, in particular when I call:

`const uint32_t ram_start = 0; // Value is not used by ble-driver
	uint32_t error_code = 0;
	ble_cfg_t ble_cfg;

	// Configure the connection roles.
	memset(&ble_cfg, 0, sizeof(ble_cfg));
	ble_cfg.gap_cfg.role_count_cfg.periph_role_count = 0;
	ble_cfg.gap_cfg.role_count_cfg.central_role_count = 1;
	ble_cfg.gap_cfg.role_count_cfg.central_sec_count = 1;

	QThread::msleep(300);
	error_code = sd_ble_cfg_set(this->adapter, BLE_GAP_CFG_ROLE_COUNT, &ble_cfg, ram_start);`

The error I get is:

`sd_ble_cfg_set() failed when attempting to set BLE_GAP_CFG_ROLE_COUNT. Error code: 0x03`

It seems a lot strange to me since the documentation for that function suggests that the return value shouldn't be 0x03.

Let me add that I currently placed a delay before calling this function (300 ms) and it seems to work perfectly all the times. If I use something smaller (like 200ms) it returns the error eventually.

Am I doing something wrong or is it a bug?

Thanks a lot!

bencefr, 2018-06-25:
Thank you for reporting, we are going to look into this.

kenr, 2018-07-12:
Hi @ptagl,

The error code you receive is NRF_ERROR_INTERNAL, which indicates that something is wrong with the communication between pc-ble-driver and the device running the connectivity firmware.

We have done some changes related to this lately, can you test with the code in the v3 branch and see if you are able to reproduce the problem now?

We will be tagging a release of v3 a bit later when we have tested a bit more.

ptagl, 2018-07-13:
I'll try to test it, but it will not be easy since my development environment is not ready to compile the driver from source; during the next days I'll try to set everything up and let you know if the problem is solved.

bihanssen, 2018-08-01:
Closing the issue due to inactivity. Please reopen if this is still an issue.

closed 2018-08-01 by bihanssen