NordicSemiconductor/pc-ble-driver #124: Failed to compile the source code

Seven-Zeng, 2018-07-14:
Hi,

I tried to compile the code but failed.

Here's the versions of the tool set:

Boost: V1.67.0
CMake: 3.12.0-rc3
Visual Studio: 2017 Community

I followed the steps illustrated at 'Installation.md', and successd to build boost, create VS project, but error occured when trying to build all with command
cmake -G "Visual Studio 15 2017 Win64" ..

Basically there are 2 errors:
- _USE is not a member of std/_USE: Cannot find identifier
- “SSCodeCvt”: Illegal to used this type as an expression (Translated from Chinese, may not so accurate)

And all the errors point to line 892 and 930 of file 'stddtring'

Did I misses anything?

Here is the error info:

`“D:\Projects\pc-ble-driver\build\ALL_BUILD.vcxproj”(默认目标) (1) ->
“D:\Projects\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj”(默认目标) (3) ->
(ClCompile 目标) ->
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C2039: “_USE”: 不是“std”的成员 [D:\Projects\pc-b
le-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C2275: “SSCodeCvt”: 将此类型用作表达式非法 [D:\Project
s\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C3861: “_USE”: 找不到标识符 [D:\Projects\pc-ble-d
river\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C2039: “_USE”: 不是“std”的成员 [D:\Projects\pc-b
le-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C2275: “SSCodeCvt”: 将此类型用作表达式非法 [D:\Project
s\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C3861: “_USE”: 找不到标识符 [D:\Projects\pc-ble-d
river\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C2039: “_USE”: 不是“std”的成员 [D:\Projects\pc-b
le-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C2275: “SSCodeCvt”: 将此类型用作表达式非法 [D:\Project
s\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C3861: “_USE”: 找不到标识符 [D:\Projects\pc-ble-d
river\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C2039: “_USE”: 不是“std”的成员 [D:\Projects\pc-b
le-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C2275: “SSCodeCvt”: 将此类型用作表达式非法 [D:\Project
s\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C3861: “_USE”: 找不到标识符 [D:\Projects\pc-ble-d
river\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C2039: “_USE”: 不是“std”的成员 [D:\Projects\pc-b
le-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C2275: “SSCodeCvt”: 将此类型用作表达式非法 [D:\Project
s\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C3861: “_USE”: 找不到标识符 [D:\Projects\pc-ble-d
river\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C2039: “_USE”: 不是“std”的成员 [D:\Projects\pc-b
le-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C2275: “SSCodeCvt”: 将此类型用作表达式非法 [D:\Project
s\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v2.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C3861: “_USE”: 找不到标识符 [D:\Projects\pc-ble-d
river\build\pc_ble_driver_obj_sd_api_v2.vcxproj]

“D:\Projects\pc-ble-driver\build\ALL_BUILD.vcxproj”(默认目标) (1) ->
“D:\Projects\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj”(默认目标) (4) ->
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C2039: “_USE”: 不是“std”的成员 [D:\Projects\pc-b
le-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C2275: “SSCodeCvt”: 将此类型用作表达式非法 [D:\Project
s\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C3861: “_USE”: 找不到标识符 [D:\Projects\pc-ble-d
river\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C2039: “_USE”: 不是“std”的成员 [D:\Projects\pc-b
le-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C2275: “SSCodeCvt”: 将此类型用作表达式非法 [D:\Project
s\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C3861: “_USE”: 找不到标识符 [D:\Projects\pc-ble-d
river\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C2039: “_USE”: 不是“std”的成员 [D:\Projects\pc-b
le-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C2275: “SSCodeCvt”: 将此类型用作表达式非法 [D:\Project
s\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C3861: “_USE”: 找不到标识符 [D:\Projects\pc-ble-d
river\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C2039: “_USE”: 不是“std”的成员 [D:\Projects\pc-b
le-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C2275: “SSCodeCvt”: 将此类型用作表达式非法 [D:\Project
s\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C3861: “_USE”: 找不到标识符 [D:\Projects\pc-ble-d
river\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C2039: “_USE”: 不是“std”的成员 [D:\Projects\pc-b
le-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C2275: “SSCodeCvt”: 将此类型用作表达式非法 [D:\Project
s\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(892): error C3861: “_USE”: 找不到标识符 [D:\Projects\pc-ble-d
river\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C2039: “_USE”: 不是“std”的成员 [D:\Projects\pc-b
le-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C2275: “SSCodeCvt”: 将此类型用作表达式非法 [D:\Project
s\pc-ble-driver\build\pc_ble_driver_obj_sd_api_v5.vcxproj]
  d:\projects\pc-ble-driver\src\common\platform\win\stdstring.h(930): error C3861: “_USE”: 找不到标识符 [D:\Projects\pc-ble-d
river\build\pc_ble_driver_obj_sd_api_v5.vcxproj]`

bencefr, 2018-07-16:
Hi @Seven-Zeng ,
Which version/branch did you try to build? This issue has been fixed recently in master and v2/v3 branches, e.g. #118 .

Seven-Zeng, 2018-07-19:
@bencefr I used the master branch

kenr, 2018-07-19:
@Seven-Zeng Do you know the MSVC compiler version you are using? If the reported version is < 1914 it uses the Microsoft non-standard facet, if equal or above the one provided by Microsofts implementation of STL.

Here is check: https://github.com/NordicSemiconductor/pc-ble-driver/blob/71b620edf09cfa7e5f4f4668b8edf105ef672922/src/common/platform/win/stdstring.h#L794

Seven-Zeng, 2018-07-25:
Hi all, sorry that I didn't realized that the code has been updated after I pulled the code, it's now been compiled successfully

closed 2018-07-25 by Seven-Zeng