NordicSemiconductor/pc-ble-driver #132: pc-ble-driver and custom UUID base on heart rate example written in Python

tstefanski, 2018-08-27:
I'm trying to use the pc-ble-driver and the python heart rate example on windows 10 to set a custom CCCD. The board that is connect to the windows PC is a PCA10040 V1.1.1

The DUT (device under test) is our own custom board with a Nordic NRF52832 running SDK 14.2 and softdevice 5.1

I have followed the example in the DevZone but when the custom UUID is read by the pc-ble-driver. it creates an bleuuid object with a uuid of 0.

This problem is the same that is described in the following link

https://devzone.nordicsemi.com/f/nordic-q-a/32759/writing-to-vendor-specific-ble-characteristics

Could you inform me if there is any progress on this issue ?

fouge, 2018-10-09:
Hello,
Same here, I can't correctly parse services with 128-bit UUID.

DTibbs, 2019-01-09:
I was running into this issue, as well, trying to discover services that used a 128bit UUID (Vendor UUID). With quite a bit of digging I found:
`SD_RPC_API uint32_t sd_ble_uuid_vs_add(adapter_t *adapter, ble_uuid128_t const *p_vs_uuid, uint8_t *p_uuid_type);`
Just call that in the application before you begin discovering services on your device that uses 128bit UUIDs. From there, you will get services with UUIDs that are 16-bit, taken from byte 12 and 13 of the 128bit UUID.

tstefanski, 2019-01-10:
Thanks for the info. I'll give it a try

Tom

bihanssen, 2019-03-19:
Closing this issue. Please use the devzone thread for follow up.

closed 2019-03-19 by bihanssen