NordicSemiconductor/pc-ble-driver #135: Source code of connectivity_2.0.1_1m_with_s132_5.0.hex

wxzed, 2018-09-03:
Is there a source code for hex running on 52832?

frvieren, 2018-10-03:
I'm wondering if the sources for the HR connectivity application provided in the examples of the SDK v15 is compatible with this BLE driver code. So is the h5 wrapping done in the transport layer of the packets decoded by the application code provided in the SDK compatible? Can it be correctly decoded, meaning stripping off the header and CRC.

frvieren, 2018-10-03:
BTW for debugging the connectivity application (for example in SEGGER IDE)  it would be great to have the sources at least if the ones provided in the SDK are not usable. So far I don't get my setup up and running, meaning having the BLE driver with the HR application running on a iMX6 Exceet evaluation board (yocto distribution) linked via UART with the Nordic nRF52 DK running the HR connectivity SW. Nor the hex provided in the BLE driver nor the one provided in the SDK v15 gives positive results. The BLE driver application stops working after trying to send for 6 times without an ACK the reset control packet. So the BLE driver gets not initialized.

bihanssen, 2018-10-08:
@wxzed, the source code for connectivity firmware already available, see the [bootstrap](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/hex/sd_api_v5/bootstrap_sd_api_v5.sh) files to get info on SDK location and patch file that needs to be applied.

Also worth noting, there is a new version of pc-ble-driver coming out soon, with modifications to build scripts: [hex folder](https://github.com/NordicSemiconductor/pc-ble-driver/tree/feature/consolidate-sd-apis/hex).

bihanssen, 2018-10-08:
@frvieren, pc-ble-driver is not compatible with the serialization examples provided with SDK (e.g. HR application), since pc-ble-driver adds some breaking but needed modifications to the serialization protocol and implementation to make it more suitable for use from PC.

closed 2018-10-08 by bihanssen