NordicSemiconductor/pc-ble-driver #143: pc-ble-driver running on iMX6 UL-2 S evk and using uart /dev/ttymxc2 as transport link

frvieren, 2018-09-26:
Hallo,

Just wondering, in the documentation it is mentioned that any architecture is supported that runs for example Linux, thus also it should work on a iMX6 UL board? Currently I'm doing a setup between a iMX6 UL-2 S Exceet evk and a Nordic nRF52 DK, the idea is to run a serialized application (e.g. the HR applic) on the iMX6 board and running the Nordic HR connectivity applic on the Nordic board. Transport link between the 2 boards is UART.

It seems a  tough job to replace the nrf_drv_uart.c driver file by one that supports the iMX6. 

Thanks,
Best regards,
Frank

frvieren, 2018-10-03:
Hi,

The nRF52 DK operating as connectivity board does not respond when the LINK CTRL packet is sent from the iMX6 exceet board. After 6 try-outs the application exists. The BLE driver is not initialized.
[image]

frvieren, 2018-10-08:
See further progress of this setup/use case in the Nordic Dev zone:

https://devzone.nordicsemi.com/f/nordic-q-a/39173/no-valid-bidirectional-uart-communication-between-the-imx6-ul-s-evk-board-running-the-nordic-ble-driver-from-git-and-the-nrf52-dk-pca10040-when-running-the-nordic-hr-connectivity-software?ReplySortBy=CreatedDate&ReplySortOrder=Descending

bihanssen, 2018-10-08:
Hi, @frvieren. I'm gonna close this issue here in favor of the DevZone ticket to avoid duplicated follow up. Feel free to re-open. Thanks.

closed 2018-10-08 by bihanssen