NordicSemiconductor/pc-ble-driver #147: sd_ble_gap_authenticate causes internal error

ntzeigem, 2018-10-10:
Here is the problem.
we're using the firmware in pc-ble-driver/hex/sd_api_v5/connectivity_2.0.1_1m_with_s132_5.0.hex on the devboard. After connecting to a peripheral it starts the authentication process with "just works" pairing to the connected peripheral. 
The BLE_GAP_EVT_SEC_PARAMS_REQUEST and BLE_GAP_EVT_CONN_SEC_UPDATE are received and handled without errors, but after that the BLE_GAP_EVT_AUTH_STATUS event causes an internal error.
The error_logger prints "Failed to decode event, error code is 9."

I reproduced this behaviour with a higher logging level and got the following output:
`Debug:      127 ->  []
               type:                 ACK reliable: no seq#:0 ack#:4 payload_length:0 data_integrity:0 err_code:0
BLE_GAP_EVT_CONN_SEC_UPDATE
Debug:   138/ 1 <-  [02 19 00 00 00 00 00 03 00 00 00 ]
               type:     VENDOR_SPECIFIC reliable:yes seq#:4 ack#:4 payload_length:b data_integrity:1 header_checksum:5e err_code:0
Debug:      128 ->  []
               type:                 ACK reliable: no seq#:0 ack#:5 payload_length:0 data_integrity:0 err_code:0
Error: Failed to decode event, error code is 9.`

I'm not sure whats going and need some help. 

Thanks in advance 
Best regards

bihanssen, 2018-10-11:
Is this on a linux or mac? If so, it could be packet length issue, in which case try to disable MSD on JLink. (related devzone issue: https://devzone.nordicsemi.com/f/nordic-q-a/29777/error-occurred-opening-serial-port-nrf_error_timeout#comment-189975)

ntzeigem, 2018-10-11:
We're on Windows, in my case Win7 and using the c driver. It's not the same error as described in the related issue you've posted. We're able to open the serial port and communicate with the devboard, meaning that we're able to do e.g. device, connect, read, write, service and characteristic discovery. 
The described error occurs only if we try to authenticate with a ble peripheral

bihanssen, 2018-10-11:
Hi, I linked to the devzone issue just because it touches upon MSDdisable, I'm aware the issue itself is not related, sorry about the confusion, I will edit and erase the link. As you are on Windows it's not relevant anyway.

bihanssen, 2019-03-19:
Closing this issue. Please report on devzone if issues are still found in the latest release.

closed 2019-03-19 by bihanssen