NordicSemiconductor/pc-ble-driver #201: Building instructions for Windows

pambardekar, 2019-02-18:
I am compiling pc-ble-driver (master) on Windows. Apart from documented build instructions, I also had to export following environment variables:
    Catch2_DIR
    ASIO_INCLUDE_DIR

Should this be added to documentation? Did anyone else also have to do this?

kwonjoon, 2019-02-18:
Hello, actually, I also had same issues.
So, I changed CMakeCache file in the "pc-ble-driver-master/build" 
in order to add path of catch 2 & asio as following.

ASIO_INCLUDE_DIR:PATH=C:\pc-ble-driver-master
Catch2_DIR:PATH=C:\vcpkg\installed\x86-windows\share\Catch2

lilliesAndRoses, 2019-02-19:
This should be a documentation bug then.

kenr, 2019-02-21:
Have you followed the instructions and installed catch2 and asio?
From the documentation: https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/README.md#compiling-pc-ble-driver-on-windows

Have you set the correct VCPKG_ROOT path and used it when invoking cmake?

From the documentation:
cmake -G "Visual Studio 14" -DCMAKE_TOOLCHAIN_FILE=**%VCPKG_ROOT%\scripts\buildsystems\vcpkg.cmake** ..

closed 2019-03-05 by kenr